contiki-ng/contiki-ng #128: RPL-lite: Slow rejoin if root resets

g-oikonomou, 2017-10-29:
1. Root starts
2. Mesh node starts and joins the network smoothly
3. Root restarts.
4. Mesh node thinks its link to the root is fresh and no longer sends DAOs. Node thinks it has joined the network, only the root has no path to it.

In the attached PCAP, the root restarts at around frame 52. See how the mesh node does not start probing with unicast DIOs until quite a few minutes later. There is still no down path to the node about 20' after the root restart.

[slow-rejoin.pcapng.zip](https://github.com/contiki-ng/contiki-ng/files/1425431/slow-rejoin.pcapng.zip)

simonduq, 2017-10-30:
Thanks for the detailed report.

The natural way to handle this is with DAG version or DTSN increments, maybe periodically or under some condition. Nodes will notice the reboot because these are lollipop, and will local repair immediately and re-send DAO.

A problem is that this will increases also in the reboot-free cases. An alternative, less portable, is to store the last version in persistent memory and increment upon reboot. Need to think more about it and measure overhead, probably for next release.

simonduq, 2018-03-25:
OK I've been giving this more thought, let me describe below the different options and their trade-offs. There are at least two dimensions in the design space: how and when to fix the topology.

How:
* No action: with no action, nodes will eventually re-register their route with a new DAO. With the current defaults in RPL Lite, this happens at a 30-minute period.
* DTSN increment: the root can increment the DTSN it sends in its DIO. When noticing the increment from their direct parent, nodes will re-register their DAO, and increment their own DTSN, asking their sub-tree to also re-register. Summary: has the whole network resend DAOs, at the pace of the current DIO Trickles, i.e., there is Trickle reset involved.
* Trickle reset on DTSN increment: we could decide, as an implementation policy, to also reset the Trickle timer whenever seeing a DTSN increment. This will result in much faster re-registration than above. But will also increase control traffic noticeably for the next tens of minutes, until Trickle converges again.
* Global repair: the root can increment the DODAG version. In reaction, nodes will perform a local repair (reset all RPL state and timers including Trickle) and propagate the new version. This results in the topology getting rebuilt from the ground up. A downside is that during the repair, a node may become temporarily parentless, as it has to attach nodes from the DAG, and may be unable to find an acceptable one (acceptable and fresh link estimates etc).

When:
* When the network operator decides. This is already available: an administrator can trigger a repair at the root via the shell.
* When no path is found towards a node. The root could assume that when a packet is sent towards a node but there is no path, the network might be in need for fix. It would then trigger one of the three methods above. A downside is that this can be triggered by malicious actors willing to increase control traffic.
* Periodically. The root could increment DTSN or perform global repairs periodically. While this may seem like a natural option, it is unclear whether it really helps given that we already have DAOs resent every 30 minutes. But perhaps this makes sense with a long time-scale, e.g. do a global repair once a day? Maybe as a config option.
* At reboot. Triggering a repair after a root reboot would make sense, but this requires storing the last DODAG version in NVM, so as to increment it before re-building a network.