contiki-ng/contiki-ng #134: Build platform Cooja on OS X

g-oikonomou, 2017-10-30:
Would be nice to be able to do this. I can provide the exact error message tomorrow, but it's basically about -dynamiclib not being used, this generates a warning which gets treated as an error and the build fails.

simonduq, 2017-10-30:
@joakimeriksson @nfi might be able to help here :)

joakimeriksson, 2017-10-31:
Yes, this would be very nice to get working on OS-X.

joakimeriksson, 2017-10-31:
let's try to get this for the Spring 2018 release (as it might be a bit of work - I have a feeling that it might be some 64 vs 32-bit java issues hidden here also - and if we fix them I guess Cooja in docker is easy also).

g-oikonomou, 2017-11-09:
Some more info:

* Firstly, this fails:

```
$ make TARGET=cooja
make: *** No rule to make target `hello-world', needed by `all'.  Stop.
```

This shouldn't be happening. There is something messed up with the platform's Makefiles somewhere. I think it could be caused by the build system starting off with building `cooja.jar`. Could we turn this into a proper prerequisite?

* Secondly

```
> make hello-world.cooja TARGET=cooja 
  CC        hello-world.c
mkdir obj_cooja
  CC        ../../arch/platform/cooja/./contiki-cooja-main.c
../../arch/platform/cooja/./contiki-cooja-main.c:38:10: fatal error: 'jni.h' file not found
#include <jni.h>
         ^~~~~~~
1 error generated.
make: *** [contiki-cooja-main.o] Error 1
Process returned error code 2
```

To fix this, we need to add this `-I/System/Library/Frameworks/JavaVM.framework/Headers` to `COMPILER_ARGS` through the external tools dialogue. It should be possible to ship with this as default on OSX.

* Thirdly: Compilation works, but then we fail to link:

```
> make hello-world.cooja TARGET=cooja 
  CC        hello-world.c
  CC        ../../arch/platform/cooja/./contiki-cooja-main.c
mv contiki-cooja-main.o obj_cooja/mtype29.o
ar rc obj_cooja/mtype29.a obj_cooja/simEnvChange.o obj_cooja/cooja_mt.o obj_cooja/cooja_mtarch.o obj_cooja/rtimer-arch.o obj_cooja/slip.o obj_cooja/watchdog.o obj_cooja/beep.o obj_cooja/button-sensor.o obj_cooja/ip.o obj_cooja/leds-arch.o obj_cooja/moteid.o obj_cooja/pir-sensor.o obj_cooja/rs232.o obj_cooja/vib-sensor.o obj_cooja/clock.o obj_cooja/cooja-log.o obj_cooja/cfs-cooja.o obj_cooja/cooja-radio.o obj_cooja/eeprom.o obj_cooja/slip-arch.o obj_cooja/random.o obj_cooja/sensors.o obj_cooja/leds.o obj_cooja/contiki-main.o obj_cooja/autostart.o obj_cooja/compower.o obj_cooja/ctimer.o obj_cooja/energest.o obj_cooja/etimer.o obj_cooja/log.o obj_cooja/mt.o obj_cooja/process.o obj_cooja/rtimer.o obj_cooja/stimer.o obj_cooja/timer.o obj_cooja/nullradio.o obj_cooja/serial-line.o obj_cooja/aes-128.o obj_cooja/assert.o obj_cooja/ccm-star.o obj_cooja/crc16.o obj_cooja/heapmem.o obj_cooja/ifft.o obj_cooja/list.o obj_cooja/memb.o obj_cooja/ringbuf.o obj_cooja/ringbufindex.o obj_cooja/trickle-timer.o obj_cooja/link-stats.o obj_cooja/linkaddr.o obj_cooja/nbr-table.o obj_cooja/net-debug.o obj_cooja/netstack.o obj_cooja/packetbuf.o obj_cooja/queuebuf.o obj_cooja/mac-sequence.o obj_cooja/mac.o obj_cooja/frame802154.o obj_cooja/frame802154e-ie.o obj_cooja/framer-802154.o obj_cooja/nullframer.o obj_cooja/csma-output.o obj_cooja/csma.o obj_cooja/ip64-addr.o obj_cooja/psock.o obj_cooja/resolv.o obj_cooja/sicslowpan.o obj_cooja/simple-udp.o obj_cooja/tcp-socket.o obj_cooja/tcpip.o obj_cooja/udp-socket.o obj_cooja/uip-debug.o obj_cooja/uip-ds6-nbr.o obj_cooja/uip-ds6-route.o obj_cooja/uip-ds6.o obj_cooja/uip-icmp6.o obj_cooja/uip-nameserver.o obj_cooja/uip-nd6.o obj_cooja/uip-packetqueue.o obj_cooja/uip-udp-packet.o obj_cooja/uip6.o obj_cooja/uipbuf.o obj_cooja/uiplib.o obj_cooja/rpl-dag-root.o obj_cooja/rpl-dag.o obj_cooja/rpl-ext-header.o obj_cooja/rpl-icmp6.o obj_cooja/rpl-mrhof.o obj_cooja/rpl-nbr-policy.o obj_cooja/rpl-neighbor.o obj_cooja/rpl-ns.o obj_cooja/rpl-of0.o obj_cooja/rpl-timers.o obj_cooja/rpl.o 
Redefining printf
objcopy --redefine-sym printf=log_printf hello-world.co;   objcopy --redefine-sym printf=log_printf obj_cooja/mtype29.o;   objcopy --redefine-sym printf=log_printf obj_cooja/mtype29.a; 
/bin/sh: objcopy: command not found
/bin/sh: objcopy: command not found
/bin/sh: objcopy: command not found
make: [obj_cooja/mtype29.cooja] Error 127 (ignored)
objcopy --redefine-sym puts=log_puts hello-world.co;   objcopy --redefine-sym puts=log_puts obj_cooja/mtype29.o;   objcopy --redefine-sym puts=log_puts obj_cooja/mtype29.a; 
/bin/sh: objcopy: command not found
/bin/sh: objcopy: command not found
/bin/sh: objcopy: command not found
make: [obj_cooja/mtype29.cooja] Error 127 (ignored)
objcopy --redefine-sym putchar=log_putchar hello-world.co;   objcopy --redefine-sym putchar=log_putchar obj_cooja/mtype29.o;   objcopy --redefine-sym putchar=log_putchar obj_cooja/mtype29.a; 
/bin/sh: objcopy: command not found
/bin/sh: objcopy: command not found
/bin/sh: objcopy: command not found
make: [obj_cooja/mtype29.cooja] Error 127 (ignored)
gcc -dynamiclib -fno-common -o obj_cooja/mtype29.cooja hello-world.co obj_cooja/mtype29.o obj_cooja/mtype29.a -framework JavaVM -Wl,-map,obj_cooja/mtype29.map
cp obj_cooja/mtype29.cooja hello-world.cooja
rm hello-world.co
```

It is possible to install binultils through homebrew. Things will download and install, but likely to fail at link stage because of clashes with msp430-binutils. Nevertheless, you have `objcopy` in a homebrew dir. It will look something like this: `/usr/local//Cellar/binutils/2.29.1/x86_64-apple-darwin17.0.0/bin/`. Stick that somewhere in the path.

* Next (which is where I got stuck)

```

> make hello-world.cooja TARGET=cooja 
  CC        hello-world.c
mkdir obj_cooja
  CC        ../../arch/platform/cooja/./contiki-cooja-main.c
mv contiki-cooja-main.o obj_cooja/mtype106.o
  CC        ../../arch/platform/cooja/lib/simEnvChange.c
[...]
Redefining printf
objcopy --redefine-sym printf=log_printf hello-world.co;   objcopy --redefine-sym printf=log_printf obj_cooja/mtype106.o;   objcopy --redefine-sym printf=log_printf obj_cooja/mtype106.a; 
[... cut ...]
objcopy --redefine-sym putchar=log_putchar hello-world.co;   objcopy --redefine-sym putchar=log_putchar obj_cooja/mtype106.o;   objcopy --redefine-sym putchar=log_putchar obj_cooja/mtype106.a; 
gcc -dynamiclib -fno-common -o obj_cooja/mtype106.cooja hello-world.co obj_cooja/mtype106.o obj_cooja/mtype106.a -framework JavaVM -Wl,-map,obj_cooja/mtype106.map
ld: object file hello-world.co was built for different x86_64 sub-type (-2147483645) than link command line (3) file 'hello-world.co' for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make: *** [obj_cooja/mtype106.cooja] Error 1
Process returned error code 2
```

g-oikonomou, 2017-11-10:
It would also be nice if Cmd+XYZ worked as copy/cut/paste on OS X, instead of having to use Ctrl+C etc :D

joakimeriksson, 2022-06-22:
Closing - as Cooja works on 64-bit ubuntu and OS-X.