contiki-ng/contiki-ng #157: Enhanced, 6lbr-friendly slip-radio

g-oikonomou, 2017-11-05:
With the extended RF API there is a number of nice things we can do with tunslip6/slip-radio.

Plus, 6lbr has had to maintain a modified version internally, so it'd be great if our slip-radio and tunslup6 get aligned with 6lbr.

joakimeriksson, 2017-11-07:
I would love to make really significant move towards a more advanced serial radio with configuration, upgrade over serial, etc. And get 6LBR to make use of that also - if possible. We have such a serial-radio in Sparrow and I am kind of hoping we could make use of it + some more features to enable TSCH scheduling in the serial radio also.

laurentderu, 2017-12-13:
The current 6LBR improvements of the slip-radio are :

- [ ] Set/get channel
- [ ] Set/get PAN-ID
- [ ] Configure MAC address
- [ ] Reset SLIP-Radio
- [ ] Serialize incoming packet attributes
- [ ] CRC-8 support (From @atiselsts)
- [ ] IP SLIP-Radio, to provide basic TSCH radio support

Should be added to be complete :
- [ ] Enable/disable radio

joakimeriksson, 2017-12-13:
We have similar list of features with CRC32 and some more dynamic stuff such as setting TX power and added a container on top of SLIP that is more generic and in the future can support the whole Yanzi application layer (for upgrading serial radio firmware). One important feature is also the multi-mode radio feature that makes it possible to switch between RSSI scanner, packet sniffer or regular serial radio at any time.

joakimeriksson, 2017-12-13:
(I am ready to help out getting this working fine with 6LBR as I need it for working for Yanzi in the future - and would avoid having multiple different SRs if possible).

simonduq, 2018-02-14:
Postponing to 4.2.

simonduq, 2018-10-19:
Marking as long-term