contiki-ng/contiki-ng #164: Copper broken by latest Firefox

zhitaoh, 2017-11-08:
Firefox upgrades its extension framework, which unfortunately stops the Copper CoAP plugin from working since version 56. And it won't be fixed in the foreseeable future because the new framework does not allow custom protocol handlers. See Matthias' explanation:
https://github.com/mkovatsc/Copper/issues/29

This means the Contiki toolchain installation wiki needs to be modified, so that users don't waste time debugging their CoAP applications using the broken Copper plugin. 

I suggest either remove the whole **Install Copper in Firefox for CoAP** paragraph, or instruct users how to freeze their Firefox at version 55, the latest version that supports Copper.

The steps I took to downgrade from v56 to v55 in Ubuntu are:

- remove current version
`sudo apt-get remove firefox`

- search for deb package of old version
https://www.google.com/search?q=firefox+55.0+14.04+deb

- download 
`cd /tmp
wget https://launchpad.net/~ubuntu-mozilla-security/+archive/ubuntu/ppa/+build/13256044/+files/firefox_55.0.2+build1-0ubuntu0.14.04.1_amd64.deb`

- install
`sudo dpkg -i firefox_55.0.2+build1-0ubuntu0.14.04.1_amd64.deb`

- freeze version from auto update
`sudo apt-mark hold firefox`

simonduq, 2017-11-08:
Thanks for reporting this and for the detailed explanation! Do you think you could propose a fix for the wiki page explaining this?

You could edit it there:
https://github.com/contiki-ng/wiki-sandbox/wiki/Toolchain-installation

And then comment again here so we integrate in the main wiki.

Thanks!

zhitaoh, 2017-11-08:
Done.

simonduq, 2017-11-09:
Awesome, thanks, now updated on main wiki:
https://github.com/contiki-ng/contiki-ng/wiki/Toolchain-installation

closing

closed 2017-11-09 by simonduq