contiki-ng/contiki-ng #168: Orchestra and fragmented packets

e120guru, 2017-11-09:
It would be really handy with a good way of scheduling a listening (and transmitting) slot immediately if the IPv6 packet was fragmented by 6lowpan layer and there is more fragments to send... This really would make a huge difference in our low duty cycle leaf nodes.

@simonduq and @joakimeriksson mentioned that this might be up for consideration in a standardisation work of Orchestra... Is, and in that case when, would that be of for grabs?

simonduq, 2017-11-10:
Well, you are spot-on. The current draft is at:
https://tools.ietf.org/html/draft-duquennoy-6tisch-asf-00

And based on the feedback at IETF99, we will add two things:
* define how to configure the schedule at bootstrap
* add support for traffic bursts, e.g. fragmented 6lowpan

The way we currently plan to support bursts is to use the IEEE 802.15.4 header "frame pending" bit to signal that there is more traffic to come. Upon receiving a unicast with this bit set, the receiving node will  automatically schedule the next slot for Rx. This goes on until a packet is received with the bit unset, or until there is a packet loss.

Labeling as `roadmap` as this is on our list!

Thanks!

simonduq, 2018-03-26:
Please take a look at https://github.com/contiki-ng/contiki-ng/pull/380 and let us know if this works for you!

PR #380 (2018-05-14): TSCH burst mode
This implements the TSCH burst mode, described in the IETF I-D ASF (https://tools.ietf.org/html/draft-duquennoy-6tisch-asf-01), and considered for inclusion in 6TiSCH. The idea is simple: for slots that support burst mode, whenever there are more packets in queue for a given neighbor, we schedule more slots on-the-fly as long as we get ACKs.

Marking as work-in-progress because I would rather get this merged after 4.1, so this gets more testing.

Builds on https://github.com/contiki-ng/contiki-ng/pull/379

Closes #168
closed 2018-05-14 by simonduq