contiki-ng/contiki-ng #186: cc2538 timer trigger insanely

yoyo890121, 2017-11-21:
Hi,
I try to add a process with timer to do something periodically in `border-router-embedded.c` and `coap-example-server.c`.
I use TSCH MAC layer and my device is cc2538dk.
I tried contiki and timer trigger insanely in beginning, and I set `LPM_CONF_ENABLE` to 0 it still occurred sometime, so I wanted to try contiki-ng.
It seemed to work well in beginning, but sometimes this situation occurred after 3 days in my test.

Thanks.

```
PROCESS(node_process, "Node Timer");

PROCESS_THREAD(node_process, ev, data)
{
  static struct etimer etaa;
  PROCESS_BEGIN();
  static int i=0;  
  etimer_set(&etaa, CLOCK_SECOND * 10);  
  while(1) {
    printf("10 seconds. \t i=%d \n", i);
    PROCESS_YIELD_UNTIL(etimer_expired(&etaa));
    etimer_reset(&etaa);
    i++;    
  }
  PROCESS_END();
}
```

simonduq, 2017-11-21:
Thanks for reporting. Would you mind sharing a full example code? (c/h and makefile).

yoyo890121, 2017-11-21:
https://gist.github.com/yoyo890121/37336870bc84d291c8f29222c4754079
https://gist.github.com/yoyo890121/6bacd2e9b1f527f2b8e76f130519d1c1

I just edited these files in `contiki-ng/examples`, and set `#define LOG_CONF_LEVEL_MAC  
  LOG_LEVEL_DBG` to print log for testing contiki-ng.

Thanks for helping.

simonduq, 2017-11-21:
OK thanks. Would be ideal if you could actually share a self-contained branch we can directly try out. BTW is the problem on the BR, the other node or both? Any steps how to reproduce and how the bug exactly manifests itself will help.

Until then, what I think you should try is a basic hello-world with your etimer loop and TSCH enabled. Please let us know how this goes!

yoyo890121, 2017-11-21:
https://github.com/yoyo890121/contiki-ng/tree/hong-testing
Hi, I forked contiki-ng and start a basic hello-world with etimer loop and TSCH enabled testing.
The problem both occurred on BR and node, and when the problem occurred, console printed 
`10 seconds.  i=(i)  ` insanely to show this situation.

Thanks.

simonduq, 2017-11-21:
Thanks, we will look into this when we find the time!

g-oikonomou, 2017-11-21:
Can you try with `etimer_restart()` instead of `_reset()`?

yoyo890121, 2017-11-21:
Yes, I started another mote to try a hello-world with `etimer_restart()` .
Thanks.

simonduq, 2017-11-24:
OK I just tried it, but on platform Zoul instead (which is also based on a cc2538 SoC). Could not reproduce the issue. It's been running for 7 minutes now, do we need to wait longer for the bug to happen?

I guess we will need someone to test on actual cc2538-dk.

Meanwhile, please let us know if you found anything.

andy199310, 2017-11-24:
Hello! Thanks for helping us!
I am co-worker of yoyo :)

I traced more code in this period and i think the problem is at **Sleep Timer**.
In this function (https://github.com/contiki-ng/contiki-ng/blob/develop/arch/cpu/cc2538/clock.c#L187)
It update the software ticks `rt_ticks_startup` by calling `RTIMER_NOW()` which refer to `rtimer_arch_now` (https://github.com/contiki-ng/contiki-ng/blob/develop/arch/cpu/cc2538/rtimer-arch.c#L115) and get `sleep timer` tick.

Because `Sleep Timer` is 32-bit and `rt_ticks_startup` is 64-bit the function split `rt_ticks_startup` into two parts and make sure if `sleep timer` wraparounds `rt_ticks_startup` will increase as we wish.
However, this function will work if `sleep timer` won't go backward.

I change part of this function to:
```
if(now < (rtimer_clock_t)prev_rt_ticks_startup) {
    cur_rt_ticks_startup_hi++;
    high_sof_rtimer = prev_rt_ticks_startup;
    high_phy_rtimer = now;
  }
```
I want to record when `sleep timer` wraparound, what's the value of software ticks `prev_rt_ticks_startup` and `sleep timer (now)`.
I got the following result:
```
high_phy_rtime: 731529382 
high_sof_rtime: 731529383 
```
Which seem like cc2538 sleep timer will go backward? 

PS: We cannot find a go way to reproduce this problem. It just happen from time to time :(

Thank you!

yoyo890121, 2017-11-24:
Hi, 
We don't know what time the bug happen, and we found that the timer jumped 131072 seconds, but the ASN number was normal.
I am trying hello-world and add code to print more clock and timer information, and wait the bug happen.

Thanks for helping!

yoyo890121, 2017-12-04:
Hi,
I tried a BR and 3 hello-world node and ran 3 days, the bug happened.
It corresponds with our observation and Andy's description.
Once the bug happen, it will happen often next time. 
The following is log of bug happened one for all.

(the bug start)
```
============================
10 seconds. 	 i=170973 
next etimer: 218845560 ticks
clock: 218845552 ticks
rtime: 189886545 ticks
[DBG : TSCH Queue] packet is added put_index 1, packet #0x200008b8
[INFO: TSCH      ] TSCH: enqueue EB packet 35 16
[INFO: TSCH      ] packet sent to 0000.0000.0000.0000, seqno 24, status 0, tx 1
[INFO: TSCH-LOG  ] {asn-0.198c8cc link-0-7-0-0 ch-12} bc-0-0 tx LL-c52f->LL-NULL, len 35, seq 24, st 0 1
[DBG : TSCH Queue] packet is added put_index 2, packet #0x200008b8
[INFO: TSCH      ] TSCH: enqueue EB packet 35 16
============================
10 seconds. 	 i=170974 
next etimer: 235623847 ticks
clock: 235623354 ticks
rtime: 190036837 ticks
[INFO: RPL       ] sending a DIS to ff02::1a
[DBG : TSCH Queue] packet is added put_index 3, packet #0x200008c8
[INFO: TSCH      ] send packet to ffff.ffff.ffff.ffff with seqno 170, queue 1 2, len 14 24
[DBG : TSCH Queue] packet is added put_index 1, packet #0x200008d8
[INFO: TSCH      ] send packet to 0012.4b00.062c.c4fa with seqno 171, queue 1 3, len 21 21
[INFO: TSCH      ] sending KA to 0012.4b00.062c.c4fa
============================
10 seconds. 	 i=170975 
next etimer: 235623847 ticks
clock: 235623360 ticks
rtime: 190038381 ticks
============================
10 seconds. 	 i=170976 
next etimer: 235623847 ticks
clock: 235623362 ticks
rtime: 190038757 ticks
[INFO: TSCH      ] packet sent to 0000.0000.0000.0000, seqno 25, status 0, tx 1
[INFO: TSCH-LOG  ] {asn-0.198c989 link-0-7-0-0 ch-16} bc-0-0 tx LL-c52f->LL-NULL, len 35, seq 25, st 0 1
============================
10 seconds. 	 i=170977 
next etimer: 235623847 ticks
clock: 235623365 ticks
rtime: 190039653 ticks
============================
10 seconds. 	 i=170978 
next etimer: 235623847 ticks
clock: 235623366 ticks
rtime: 190040018 ticks
============================
```
(timer trigger insanely)
```
============================
10 seconds. 	 i=184094 
next etimer: 235642896 ticks
clock: 235642552 ticks
rtime: 194951497 ticks
============================
10 seconds. 	 i=184095 
next etimer: 235642896 ticks
clock: 235642553 ticks
rtime: 194951861 ticks
[DBG : TSCH Queue] packet is added put_index 4, packet #0x200008b8
[INFO: TSCH      ] TSCH: enqueue EB packet 35 16
[INFO: TSCH      ] packet sent to 0000.0000.0000.0000, seqno 72, status 0, tx 1
[INFO: TSCH-LOG  ] {asn-0.199052c link-0-7-0-0 ch-12} bc-0-0 tx LL-c52f->LL-NULL, len 35, seq 72, st 0 1
============================
10 seconds. 	 i=184096 
next etimer: 235643352 ticks
clock: 235642992 ticks
rtime: 195064020 ticks
[DBG : TSCH Queue] packet is added put_index 5, packet #0x200008b8
[INFO: TSCH      ] TSCH: enqueue EB packet 35 16
[INFO: TSCH      ] packet sent to 0000.0000.0000.0000, seqno 73, status 0, tx 1
[INFO: TSCH-LOG  ] {asn-0.1990691 link-0-7-0-0 ch-16} bc-0-0 tx LL-c52f->LL-NULL, len 35, seq 73, st 0 1
[INFO: TSCH-LOG  ] {asn-0.1990732 link-0-7-0-0 ch-20} drift -186
[INFO: TSCH-LOG  ] {asn-0.1990732 link-0-7-0-0 ch-20} bc-0-0 rx LL-c4fa->LL-NULL, len 35, seq 251, dr -1, edr 1
[DBG : TSCH Queue] packet is added put_index 6, packet #0x200008b8
[INFO: TSCH      ] TSCH: enqueue EB packet 35 16
[INFO: TSCH      ] packet sent to 0000.0000.0000.0000, seqno 74, status 0, tx 1
[INFO: TSCH-LOG  ] {asn-0.1990804 link-0-7-0-0 ch-12} bc-0-0 tx LL-c52f->LL-NULL, len 35, seq 74, st 0 1
[INFO: RPL       ] sending a DIS to ff02::1a
[DBG : TSCH Queue] packet is added put_index 0, packet #0x200008b8
[INFO: TSCH      ] send packet to ffff.ffff.ffff.ffff with seqno 171, queue 1 1, len 14 24
[INFO: TSCH      ] packet sent to 0000.0000.0000.0000, seqno 75, status 0, tx 1
[INFO: TSCH-LOG  ] {asn-0.1990923 link-0-7-0-0 ch-24} bc-1-0 tx LL-c52f->LL-NULL, len 24, seq 75, st 0 1
============================
```
(the bug end)

And once the bug happened, the TSCH time drift more and more.
```
[INFO: TSCH-LOG  ] {asn-0.1990732 link-0-7-0-0 ch-20} drift -186
[INFO: TSCH-LOG  ] {asn-0.1990c3a link-0-7-0-0 ch-20} drift -187
```
But we still cannot find a way to reproduce this problem.  :<

Thank you!

simonduq, 2017-12-17:
Hi, any news on this? Any luck fixing or reproducing the bug?

yoyo890121, 2017-12-18:
Hi, we modified `cc2538/clock.c` `update_ticks()` to bypass this clock backward issue.
This fixing is OK in our recent testing.
But we can not reproduce the bug, just bypass it.
Thank you!
```
  update_counter++;

  now = RTIMER_NOW();
  prev_rt_ticks_startup = rt_ticks_startup;

  cur_rt_ticks_startup_hi = prev_rt_ticks_startup >> 32;
  if(now < (rtimer_clock_t)prev_rt_ticks_startup) {
    // prevent now going backward
    if((rtimer_clock_t)prev_rt_ticks_startup - now > 2) {
      high_last_update_done_rtimer = last_update_done_rtimer;
      cur_rt_ticks_startup_hi++;
         
      high_sof_rtimer = prev_rt_ticks_startup;
      high_phy_rtimer = now;
    }else {
      strange_happen_counter++;
    }
  }

  last_update_done_rtimer = rt_ticks_startup;
```

simonduq, 2017-12-18:
OK! Just checking, is this not just a timer wrap? rtimer is unsigned and is expected to wrap. For rtimer clock comparison across wraps, use `RTIMER_CLOCK_LT`.
Was this actually the problem?

yoyo890121, 2017-12-18:
Hi, we think the problem is Andy said before.
When software rtimer is faster than physical timer, the problem will occur in original code.
```
clock: 23663 ticks
NextT: 23535 ticks
phy_rtime   : 6057805 
sof_rtime   : 6057796 
last_update : 6057541 
epoch    : 6057817 
update count: 28424 
high_phy_rtime: 0 
high_sof_rtime: 0 
high_last_upda: 0 
================
clock: 16800894 ticks
NextT: 23663 ticks
phy_rtime   : 6061647 
sof_rtime   : 4301028926 
last_update : 6061632 
epoch    : 4301028947 
update count: 28442 
high_phy_rtime: 6061630 
high_sof_rtime: 6061632 
high_last_upda: 6061375 
================
```

||hi|low|
|---|---|---|
|SW timer (64 bits)|100|999|
|HW timer (32 bits)|(100)|900|
`update_ticks() cur_rt_ticks_startup_hi++;`
|SW timer (64 bits)|102|100|
|HW timer (32 bits)|(101)|100|

So we add this line to prevent this situation.
`if((rtimer_clock_t)prev_rt_ticks_startup - now > 2)`
But we don't know why software rtimer run faster than physical timer.
Thank you.

g-oikonomou, 2018-05-25:
Just revisiting this. @yoyo890121 @andy199310 if I understand correctly you are saying that sometimes this conditional:

```
  if(now < (rtimer_clock_t)prev_rt_ticks_startup) {
```
is true, even when the rtimer has not wrapped around.

One thing I'd like to check is the caller to `update_ticks()`. It might help to know if this happens on device wakeup or on a normal SysTick interrupt.

I'm also wondering whether adding the line below to the start of `rtimer_arch_now()` makes any difference.

```
  while(!(REG(SYS_CTRL_CLOCK_STA) & SYS_CTRL_CLOCK_STA_SYNC_32K));
```

g-oikonomou, 2018-05-25:
One more thing to try out:

```
$ git diff -- arch/cpu/cc2538/lpm.c
diff --git a/arch/cpu/cc2538/lpm.c b/arch/cpu/cc2538/lpm.c
index 68cb42a..79612e8 100644
--- a/arch/cpu/cc2538/lpm.c
+++ b/arch/cpu/cc2538/lpm.c
@@ -40,6 +40,7 @@
 #include "sys/process.h"
 #include "dev/sys-ctrl.h"
 #include "dev/rfcore-xreg.h"
+#include "sys/int-master.h"
 #include "rtimer-arch.h"
 #include "lpm.h"
 #include "cc2538_cm3.h"
@@ -220,6 +221,8 @@ lpm_exit()
 
   /* Restore PMCTL to PM0 for next pass */
   REG(SYS_CTRL_PMCTL) = SYS_CTRL_PMCTL_PM0;
+
+  int_master_enable();
 }
 /*---------------------------------------------------------------------------*/
 void
@@ -319,6 +322,7 @@ lpm_enter()
     } else {
       ENERGEST_SWITCH(ENERGEST_TYPE_CPU, ENERGEST_TYPE_DEEP_LPM);
     }
+    int_master_read_and_disable();
     assert_wfi();
   }
```

lukasostendorf, 2018-11-09:
Hey,

I think I am encountering the same issue, it is just as @g-oikonomou concluded.
I am running a test in which I send a constant data-stream over multiple hops using tsch-mac and nullnet.
The tests last multiple hours, every 2 hours I change some parameters using an etimer.

A slight difference is that in my app, the difference between the `now` and `prev_rt_ticks_startup` variable is not just 1 or 2 ticks, it is in the range of 50-200.

Have a look at at the [beginning of update_ticks()](https://github.com/contiki-ng/contiki-ng/blob/develop/arch/cpu/cc2538/clock.c#L193):
```c
now = RTIMER_NOW();
prev_rt_ticks_startup = rt_ticks_startup;
```

Could the reason why `now` is a little bit smaller than `prev_rt_ticks ` be that the function is interrupted immediately after the `now` variable is set and the interrupt then changes `rt_ticks_startup`?
Therefore after the interrupt, `prev_rt_ticks_startup` is set to the updated (and therefore larger than `now`) `rt_ticks_startup`.

As suggested, I logged the caller function when update_ticks behaves strangely. It is `clock_adjust()`, so it looks like the error happens on device wakeup.
Is it possible that while `clock_adjust()` is executing update_ticks(), `clock_isr()` interrupts update_ticks?

Disabling interrupts while setting the two variables seems to fix the issue:
```c
INTERRUPT_DISABLE();
now = RTIMER_NOW();
prev_rt_ticks_startup = rt_ticks_startup;
INTERRUPT_ENABLE();
```

I ran this changed code for 15h on two devices by now and the problem did not occur again.

Another possible fix should be setting the `prev_rt_ticks` variable first, so an interrupt between the operations will have no effect:
```c
prev_rt_ticks_startup = rt_ticks_startup;
now = RTIMER_NOW();
```

Any thoughts on this, or something else I could test to identify the cause of this behavior?

simonduq, 2018-11-12:
Just wanted to add here that I believe I just witnessed the issue in a nightly testbed runs: https://contiki-ng.github.io/runs/5255.html

At after 206 minutes, at node 7, the timer starts going crazy. 9 minutes later, it expires the DAG as a result and leaves the network. The timer seems to eventually go back to normal after a while, without any reboot.

g-oikonomou, 2018-11-12:
I'm afraid I don't have the ability to try this out, but if there is a patch + explanation, perhaps we should open a pull and run it on the nightly to see what happens?

simonduq, 2018-11-12:
Good idea I'll add it to the nightly as soon as we have a PR :)

prodomo, 2019-05-02:
Hi, I'm not sure I encountering the same issue or not.
I  am running a test which periodic sending packet every 10 sec using zoul.
I print out all of LOG and save it.
I found that the last tx time in RPL nbr list would be strange.
It changed to  a large number sometimes but It was a small number as 1 or 2 min only few second ago.

In normal it would show like below
### [INFO: RPL       ] received a multicast-DIO from fe80::212:4b00:18e6:9d76, instance_id 0, DAG ID fd00::212:4b00:18e6:9ce1, version 240, dtsn 240, rank 404
[INFO: RPL       ] nbr: own state, addr fd00::212:4b00:18e6:9c63, DAG state: reachable, MOP 1 OCP 1 rank 421 max-rank 1313, dioint 14, nbr count 9 (Periodic)
[INFO: RPL       ] nbr: fe80::212:4b00:18e6:9cf7  499,   128 =>   627 --  8   a   (last tx 13 min ago)
[INFO: RPL       ] nbr: fe80::212:4b00:18e6:9ce1  128,   293 =>   421 -- 16 rbafp (last tx 0 min ago)
[INFO: RPL       ] nbr: fe80::212:4b00:18e6:9c6a  262,   341 =>   603 --  1   a   (last tx 12 min ago)
[INFO: RPL       ] nbr: fe80::212:4b00:18e6:9c8d  265,   211 =>   476 --  3   a   (last tx 8 min ago)
[INFO: RPL       ] nbr: fe80::212:4b00:14b5:d9be65535,   316 => 65535 --  3       (last tx 7 min ago)
[INFO: RPL       ] nbr: fe80::212:4b00:18e6:9dcc  504,   136 =>   640 --  3   a   (last tx 2 min ago)
[INFO: RPL       ] nbr: fe80::212:4b00:14b5:dea7  283,   248 =>   531 --  3   a   (last tx 0 min ago)
[INFO: RPL       ] nbr: fe80::212:4b00:18e6:9d76  404,   186 =>   590 --  3   a   (last tx 4 min ago)
[INFO: RPL       ] nbr: fe80::212:4b00:18e6:9b70  391,   206 =>   597 --  3   a   (last tx 5 min ago)
[INFO: RPL       ] nbr: end of list

When strange behave happened, it would print like below.

### [INFO: RPL       ] nbr: own state, addr fd00::212:4b00:18e6:9c63, DAG state: reachable, MOP 1 OCP 1 rank 400 max-rank 1313, dioint 13, nbr count 9 (Periodic)
[INFO: RPL       ] nbr: fe80::212:4b00:18e6:9cf7  478,   128 =>   606 --  8   a   (last tx 2198 min ago)
[INFO: RPL       ] nbr: fe80::212:4b00:18e6:9ce1  128,   272 =>   400 -- 16 rba p (last tx 2184 min ago)
[INFO: RPL       ] nbr: fe80::212:4b00:18e6:9c6a  268,   341 =>   609 --  1   a   (last tx 2197 min ago)
[INFO: RPL       ] nbr: fe80::212:4b00:18e6:9c8d  309,   211 =>   520 --  3   a   (last tx 2193 min ago)
[INFO: RPL       ] nbr: fe80::212:4b00:14b5:d9be65535,   316 => 65535 --  3       (last tx 2192 min ago)
[INFO: RPL       ] nbr: fe80::212:4b00:18e6:9dcc  524,   136 =>   660 --  3   a   (last tx 2187 min ago)
[INFO: RPL       ] nbr: fe80::212:4b00:14b5:dea7  268,   248 =>   516 --  3   a   (last tx 2185 min ago)
[INFO: RPL       ] nbr: fe80::212:4b00:18e6:9d76  400,   186 =>   586 --  3   a   (last tx 2189 min ago)
[INFO: RPL       ] nbr: fe80::212:4b00:18e6:9b70  355,   206 =>   561 --  3   a   (last tx 2190 min ago)
[INFO: RPL       ] nbr: end of list
[INFO[DBG : Frame 15.4] ie type 0, current state 0
[DBG : Frame 15.4] header ie: len 2 id 1e
: RPL       ] probing fe80::212:4b00:18e6:9ce1 (urgent) last tx 2184 min ago
[INFO: RPL       ] sending a unicast-DIO with rank 400 to fe80::212:4b00:18e6:9ce1

kyh-ly, 2021-03-24:
Hi， I used the CC2538 sleep timer function, and I find the program stop at the location :while(!(HWREG(SMWDTHROSC_STLOAD) & SMWDTHROSC_STLOAD_STLOAD));
The watch value shows:
SMWDTHROSC_STLOAD_STLOAD	Error (col 1): Unknown or ambiguous symbol. SMWDTHROSC_STLOAD_STLOAD			
SMWDTHROSC_STLOAD	0	0x400d5050	unsigned long