contiki-ng/contiki-ng #189: Noise generator example

atiselsts, 2017-11-22:
I noticed there are no examples that show how to create radio noise, and not many / limited examples that show how to use the `NETSTACK_RADIO` interface directly. Both of which I think are important if Contiki-NG is to remain as research-friendly OS. 

I have some code for a simple noise generator, what's your view on adding this to examples? It's useful especially for Cooja simulations; in fact, in Bristol we use it for our SPHERE regression tests.

There are Disturber motes in Cooja, but they do not play well with multichannel simulations - it's not possible to generate interference on just a specific range of channels.

simonduq, 2017-11-22:
These definitely sound like useful features!

mbaddeley, 2017-11-22:
I'd quite like this!