contiki-ng/contiki-ng #191: SLIP-cmd ignores UIP_LLH_LEN

g-oikonomou, 2017-11-22:
To send an IPv6 datagram over SLIP, what we should be doing (and have been, even before #190) is to start at `&uip_buf[UIP_LLH_LEN]`. However, we still send `uip_len` bytes, which is clearly wrong. The current code base works by fluke, because on all cases where SLIP is being used `UIP_LLH_LEN` is 0. Change `UIP_LLH_LEN` and we'll overrun `uip_buf`. An easy fix could be to send `uip_len - UIP_LLH_LEN`.

Secondly, the SLIP-cmd interface puts commands/responses into `uip_buf` starting at position 0 and then uses slip_send. Things only work because `UIP_LLH_LEN` is 0. An easy fix could be to start populating/parsing starting from position `UIP_LLH_LEN`

joakimeriksson, 2017-11-22:
Another bug is that the slip-cmd module is named slip-cmd - it is defaulting on a function related to slip but it is from what I can see not really doing any slip at all. Just sending and receiving from any buffer to a specified output function. We should probably consider another name for this module/service.

In the slip-radio LLH is always zero as there is no IP stack there - just uip-buf  - and not much else. On the border router side I am not sure it is ever hooked on the IP stack either?! We should probably rename a lot of the slip-related stuff that was initially (10-15 year ago) possibly interesting for hooking up IPv4 via SLIP frames as there were tools for that. Now I am not sure we are using it more than for basic framing over serial line - and this slip-cmd service is not even just for that case.

g-oikonomou, 2017-11-22:
Fully agree Joakim. Perhaps to quickly fix we just need to explicitly set `UIP_CONF_LLH_LEN` to 0 in the slip example.

My thinking is that we currently have 3 things:

* SLIP
* a request/response protocol over a serial type interface
* debug I/O

I've been kicking myself about a generalised char I/O design and this is exactly what I'm working on (Really, #190 is part of it - tidy up before reengineering). Currently the above three are hard-coded to be over the same physical interface: There is space for improvement there. The one thing I'm still pondering is the feature we have currently whereby a debug print statement can be interleaved in the middle of a SLIP frame. I want to keep this one.

simonduq, 2018-04-05:
Just noticed this was tagged 4.1 but not assigned to anyone. Can we realistically fix this by 4.1?

joakimeriksson, 2018-05-04:
This is now fixed via PR #462

PR #462 (2018-05-04): Updated users of the SLIP driver to respect UIP_LLH_LEN
The SLIP driver adds incoming data into `uipbuf[UIP_LLH_LEN]` and some users of the SLIP driver simply assumed that UIP_LLH_LEN is zero. This PR makes sure that listeners on the SLIP driver respect `UIP_LLH_LEN` and this fixes #191.
closed 2018-05-04 by joakimeriksson