contiki-ng/contiki-ng #197: Semantics of the Radio API

g-oikonomou, 2017-11-23:
The radio API is not documented and this has caused confusion as to how functions are meant to behave. I think we need to document `radio.h` in terms of what each function is meant to do:

* `init`:
  * Is it meant to enable frame filtering and auto ACKs, if supported? Will TSCH then go and disable those things since they need done in software?
  * Is it meant to select a default channel, or will the channel get populated explicitly later?
  * Is it meant to put the radio in RX?
* `transmit`, `channel_clear`, `send`, `receiving_packet`:
  * If they are called when the radio is off, are they expected to turn the radio on automatically, or are they expected to fail?
  * If they are expected to turn the radio on automatically, what are they expected to do at the end of the operation? Restore the radio to its previous state? Leave the radio on?
  * Is the MAC layer perhaps guaranteed to explicitly turn the radio on before it tries to perform one of those operations (so the answer to the previous question would be to fail if off)?

simonduq, 2017-11-24:
We really need to fix this. A lot easier now with our more limited set of supported platforms.
One more:
* `RADIO_RX_MODE_ADDRESS_FILTER`: is this only about address filtering (as the name suggests) or is it about 802.15.4 frame parsing (as some of the radio drivers implement)? Shall we add `RADIO_RX_MODE_FRAME_FILTER`?

g-oikonomou, 2019-06-09:
Hi there all.

I have been looking at this extensively over the last few days. Our MAC layers are being a little inconsistent in terms of how they use the radio API. This could be a result of our MAC layer implementations second-guessing what the radio does, which in turn has resulted in radio drivers second-guessing what the MAC layer does.

## The behaviour of the `transmit()` function
The main inconsistency is in terms of the `transmit()` function:

* CSMA will call `on()` once at the start and it will then call `transmit()` whenever it has a frame to send. However, it will not explicitly put the radio back in RX after TX - it will expect this transition to happen automatically. Now this is fine for radios that automatically revert to RX after TX, but not all radios do (especially so sub-ghz radios). Assuming this behaviour, our drivers try to put radios back in RX after TX.
* TSCH behaves slightly differently:
  * If we use `TSCH_RADIO_ON_DURING_TIMESLOT` then the behaviour is similar to CSMA. However if `TSCH_RADIO_ON_DURING_TIMESLOT` is 0 then TSCH will explicitly call `on()` after TX.
  * If we want CCA before TX, then TSCH will explicitly call `on()` at some point before `transmit()`. If not, then it won't.

As a result of that, we have various scenarios for `transmit()`: It may get called while the radio is powered down, or it may be called while the radio is in RX. It may be expected to revert to RX, but sometimes this is unnecessary since it will be followed by a call to `off()`. Or, if the radio was off, it may try to turn it back off. This has led to most implementations being quite convoluted.

## CCA before TX
The second inconsistency is in terms of whether we perform CCA before TX. CSMA doesn't care and leaves it to the radio driver to determine whether to do this or not. We do have `RADIO_TX_MODE_SEND_ON_CCA`, but CSMA does not make any attempt of manipulating that. Some drivers support performing CCA before TX, some don't. TSCH on the other hand explicitly turns this off at the radio driver. If it is configured with CCA before TX, it will do so at software by a sequence of calls to `on()` and `channel_clear()`.

The result of that further complexity in `transmit()` implementations.

IMHO we should just get rid of `RADIO_TX_MODE_SEND_ON_CCA` from all drivers and then implement it conditionally in software for all MACs, same as what TSCH does.

## Address filtering
Correct me if I am wrong here:

* We want to be able to disable auto ACKs for TSCH and sniffers.
* We want to be able to disable address-based frame filtering surely for sniffers, but do we also need to do this for TSCH? If so, why? I vaguely remember some discussion about frame versions supported by radios, but I am not sure now. Do we have any radios that don't support v2?

In the case of a radio that does support address-based frame filtering and v2 frames, what we should be doing is enable filtering but disable auto ACKs. This is possible for all radios that I am familiar with, but I worry that perhaps frame filtering might mess up TSCH operation? I am thinking a situation where `receiving_packet()` returns true, but then the packet gets subsequently discarded. Does our TSCH implementation handle this? If so, frame filtering is good - there is a lot of junk flying around!

## Proposal
Here is what I am proposing to improve the situation without making any changes to the API. We will almost certainly have to modify some radio drivers and also CSMA. Perhaps some changes required for TSCH here and there, but not 100% certain.

* `transmit()`: should be kept as simple as possible:
  * It should not try to guess what the MAC layer may need to do next. It may, but should not be expected to put the radio in RX after TX. What I am proposing is that the MAC layer make no assumptions about the state of the radio after TX. If it wants to revert to RX it should explicitly call `on()`. If it needs to power down the radio it should call `off()`. In the case of some drivers this will prevent unnecessary time spent in transition to RX.
  * It should expect to be called from a powered down state. In this case it should power up and go from the disabled state directly to TX, bypassing the RX state. The only exception is if the driver has to first perform CCA, unless we get rid of this as above.
* `on()` should be conceptually equivalent to "Do everything required to enter RX from any state". If the radio is on but idle it should simply perform a transition to RX. If powered down, it should first power on. The radio driver will not expect a call to `on()` before a call to `transmit()`.

Furthermore, I am proposing clarifications about the following:

* `init()`: Should only bother with one-off initialisation of the radio. Anything that does not persist across on/off cycles should be done in `on()` instead. `init()` may, but does not have to put the radio in RX mode.
* `off()` must power down the radio, but must not discard any previously received frames
* `on()` must not discard any previously received frames
* `channel_clear()` must be preceded by a call to `on()`, or fail with default return value if called while the radio is not in RX.
* `receiving_packet()` must be preceded by a call to `on()`, or simply return 0 without making any attempt of entering RX.
* `pending_packet()` may be called while the radio is powered down. In this case, the driver must be able to return the correct value _without_ manipulating the radio hardware.
* `read()`: Ditto
* `prepare()` may be called while the radio is powered down.

## Extended Documentation
I am proposing updated API documentation in `radio.h` based on the above.

Here it is on readthedocs: https://g-oikonomou-contiki-ng.readthedocs.io/en/contrib-radio-doc/_api/group__radio.html

And the raw file: https://github.com/g-oikonomou/contiki-ng/blob/contrib/radio-doc/os/dev/radio.h

I do have some more queries but they can wait for now.

yatch, 2019-06-10:
+1 to clarifications on the radio APIs!

g-oikonomou, 2020-10-22:
Updated API documentation in #1422 and changing the title here. Keeping open so it doesn't fall through the cracks.