contiki-ng/contiki-ng #207: CoAP not parsing options

AntonioLangiu, 2017-11-28:
I noticed that the CoAP implementation is not parsing the received options.

After some debug I found out that during the refactor of er-coap [this line](https://github.com/contiki-ng/contiki-ng/blob/develop/os/net/app-layer/coap/coap.c#L547) was added in the`coap_parse_message` function. Removing that line fixes the problem.

joakimeriksson, 2017-11-28:
Thanks for the report. That certainly looks like a bug. We are currently working on a big PR with CoAP, LWM2M and DTLS and we will ensure that this bug is fixed when that gets in (or even before as a separate fix-PR).

simonduq, 2017-11-29:
addressed in https://github.com/contiki-ng/contiki-ng/pull/210

joakimeriksson, 2017-11-30:
and now #210 is in - @AntonioLangiu: I guess this can be closed ?

AntonioLangiu, 2017-11-30:
Yes, it can be closed! Thank you for the fast merge!

closed 2017-11-30 by AntonioLangiu