contiki-ng/contiki-ng #211: Convergence with 6lbr

g-oikonomou, 2017-11-30:
There is an intention to converge Contiki-NG with 6lbr, in view of one including the other as a submodule in the future.

6lbr should not have to maintain modifications to the Contiki-NG network stack, it is much better for any changes required by 6lbr to be merged into Contiki-NG altogether.

This issue is to maintain a list of changes required so the two projects can play nicely with each other. I can immediately think of:

* [ ] Extensions to the command/response protocol used by slip-radio
* [ ] Ability to change PAN ID at runtime (I believe `frame802154_set_pan_id()` is not sufficient, we also need to update radio registers so that frame filtering can keep working correctly).

Ping @laurentderu

joakimeriksson, 2017-11-30:
Extensions we have at Yanzi:
* All sorts of commands over the "great" !/? protocol including setting beacons, sending "timed" packets, etc.
* A more generic set-radio-parameter command that can be used via the regular generic radio parameter API:s (param, value)
* full serialisation of all packet attributes (including INTERFACE_ID)
* Encapsulation on top of slip - with CRC32 on the data to avoid sending out crap in air or putting in crap into the IP-stack.
* Full support for the Yanzi application layer (TLVs) so that it is possible to read out data over that protocol also and write data over that protocol. This makes it possible to support over-the-serial upgrades of the serial-radio.
* Dual-mode serial-radio that when it is not connected to USB can act as a routing node (e.g. if powered by batteries or USB charger)
* Multi-mode radio interface that can do sniffing, RSSI scanning on all channels - or regular serial-radio mode.

laurentderu, 2017-12-13:
The first wave of bug fixes from 6LBR is almost done, here are the next steps I propose to converge on the code (the lists are non-exhaustive) :

- [ ] Update SLIP-Radio with 6LBR improvements (#157) (Having Yanzi slip-radio in 6LBR would be great but, as it requires much more effort, should be done in a later phase IMHO)
- [ ] Add hook or layer between tcpip.c and sicslowpan for the Packet Forwarding Engine (TODO)
- [ ] Replace activation-by-macro of some features by conditional code (TODO)
- [ ] Add hook in RPL to be notified when DODAG version changes (TODO)
- [ ] Fix multi address support (TODO)
- [ ] Hooks in ND6 to configure RA, RS, NUD behaviour (TODO)
- [ ] Hooks in tcpip to customize next-hop selection (TODO)

Next the following features could be integrated into Contiki-NG :
- [ ] Route Information Option (RIO)
- [ ] MLDv1 support (Required for the Ethernet side of Multicast)
- [ ] Multi-Radio support (MAC address learning, packet switching, ...)
- [ ] mDNS
- [ ] ND-Proxy (Required for "SmartBridge" mode)

And the second wave of (minor) bug fixes :)

simonduq, 2018-03-23:
@laurentderu any update on this? Anything we can do on our side to speed-up the process?

simonduq, 2018-09-28:
Removed the milestone due to lack of activity. This remains on the long-term roadmap

joakimeriksson, 2025-03-19:
6lbr seems relatively inactive now so I will close this issue.

closed 2025-03-19 by joakimeriksson