contiki-ng/contiki-ng #214: Time synchronization and data timestamping: documentation and example code

atiselsts, 2017-12-01:
A simple, straightforward to use time synchronization demo is something that Contiki-NG is still missing - judging by the amount of Contiki-related questions I keep seeing about it.

Something like this:
* Is based on TSCH - assumes single, synchronized global clock for the whole network;
* Either uses TSCH timeslot granularity (no core TSCH changes needed) or `rtimer` tick granularity (some core changes needed) ;
* Assumes negligible delay for processing and over-the-serial data transfer;
* Shows how the sensor network time can be converted to an UNIX timestamp on the host side.

My experience is that 1/100 second timestamps are more than enough for most applications. What do you think?

simonduq, 2017-12-02:
I fully agree on the need. I would say tick granularity rather than slot. One challenge: applications clocks usually require monotonic increase, i.e. the clock must never stop nor go back in time. And this is a property that we do not have to enforce at the TSCH level, i.e., if a node is early it will set its time instantly back in time.

This is something to consider if we want to expose a general-purpose clock.

atiselsts, 2017-12-04:
I would prefer timeslot granularity (10 ms). However, if there are specific applications that require better accuracy, and are of interest to many Contiki-NG users then obviously we need to consider that. I just wonder what those applications are?

The tick granularity basically opens a can of worms.
* what you mentioned - lack of monotonicity.
* interoperability between different platforms - rtimer tick frequency is more hardware dependent. If two platforms can talk using TSCH, they have (roughly) the same timeslot size, but they may have different notion of ticks.
* need for a 64-bit counter, which might be wasteful, especially if included in network packets. 64 bits are needed as when using a 32-bit counter and 1/32768 second ticks, overflow happens in every 36 hours, which is way too little for most applications.  But when using 10-ms timestamps, overflow only happens in 497 days, which sufficient for most applications. For the apps that need more, it's easier to disambiguate between a reboot and an overflow if they happen so rarely.
* in contrast, in some cases the 10-ms timestamp can be removed from the network packet altogether, as an optimization (e.g. sampling rate done by TSCH schedule and no packet retransmissions).
* the time for processing and the time for over-the-serial transfer cannot be assumed to be constant or negligible compared to the duration of the tick, but is most cases are much shorter than 10ms.

simonduq, 2018-03-26:
I just wanted to point to this IETF I-D that describes a global timesync mechanism for 6TiSCH: https://tools.ietf.org/html/draft-vilajosana-6tisch-globaltime-00.html

atiselsts, 2018-03-26:
Might be nice future extension. I see that the draft is still in early stages, so would not rush it right now.

simonduq, 2018-03-26:
Agree. Shall we flag this issue as "roadmap" btw? Any particular milestone or "long term"?

atiselsts, 2019-04-16:
I think this should be closed now!

closed 2019-04-16 by atiselsts