contiki-ng/contiki-ng #217: Travis Cron Jobs

simonduq, 2017-12-02:
Run periodic jobs on `develop`, testing more than what we have on PRs. See:
https://docs.travis-ci.com/user/cron-jobs/

Current plan:
* Platform-independent examples: compile all to all platforms. We will need a list of exceptions, to not run into memory overflow on sky.
* Platform-specific examples: compile all of them.
* Simulation tests: run all tests based on `Makefile.simulation` with multiple seeds (set `RUNCOUNT`). Explore different seeds at each run (set `BASESEED`). The current scripts already show at the end which seeds fail. NB: Travis builds have a maximum duration of about 50 minutes.
* Reporting: we need a way to report failed tests because we won't be logging in Travis to check. Open a GitHub issue? Email? Something else?

simonduq, 2018-04-05:
Any update on this @atiselsts ?

simonduq, 2018-04-26:
Postponing to 4.2

simonduq, 2018-09-28:
I'll look into this

simonduq, 2018-10-16:
OK, so I took a closer look, and this is not as straightforward as I thought. Basically one needs to add more jobs to the travis build, some for the Cron case, some for the normal case. But all jobs will spawn a VM at every single bulid, and only after the VM is ready be able to detect that it's not supposed to run now before exiting. Adds large overhead to every single build, so, not desirable.

Removing the milestone until we come up with a neat way to achieve this.

atiselsts, 2020-11-26:
We should probably close this as we're not using Travis anymore. What do you think?

g-oikonomou, 2020-11-26:
If there is need to run things not covered by our CI action (e.g. build all?), we can create separate actions that don't run on PRs.

Github actions has a cron syntax by the way

https://docs.github.com/en/free-pro-team@latest/actions/reference/workflow-syntax-for-github-actions#onschedule