contiki-ng/contiki-ng #220: Asynchronous RF operations

g-oikonomou, 2017-12-03:
Currently all our RF operations are synchronous.

Especially for TX, we block waiting for potentially a very long time. It would be nice to be able to return immediately at the start of the TX operation, with the radio notifying about the end of TX through a callback.

A similar technique could be applied to the `on()` operation, in the case of radios that have a very long startup time, e.g. due to a need to fire up a crystal.

arurke, 2017-12-03:
:+1: 
For what is worth, I remember @bkozak-scanimetrics implemented an async radio driver for cc2650 as part of https://github.com/contiki-os/contiki/pull/1549 . I never got around to testing it myself though.

yatch, 2017-12-04:
Interesting; this would need some adjustments on MAC implementations including TSCH, wouldn't it?

g-oikonomou, 2017-12-04:
Yep, MAC implementations as well as radio drivers to get those TX notifications.

yatch, 2017-12-04:
@g-oikonomou OK; thanks! 😄