contiki-ng/contiki-ng #244: LLSEC is broken for CSMA

joakimeriksson, 2017-12-12:
There is no longer any way to use LLSEC for CSMA and anything other than TSCH. And the framer still seems to have code that make use of attributes that are removed from packetbuf:
```
#if LLSEC802154_USES_FRAME_COUNTER
  params->aux_hdr.frame_counter.u16[0] = get_attr(PACKETBUF_ATTR_FRAME_COUNTER_BYTES_0_1);
  params->aux_hdr.frame_counter.u16[1] = get_attr(PACKETBUF_ATTR_FRAME_COUNTER_BYTES_2_3);
#else /* LLSEC802154_USES_FRAME_COUNTER */
```

We need to have support for 802.15.4 link layer security for CSMA also as this is used by some commercial Contiki users.

joakimeriksson, 2017-12-13:
I have a working Contiki-NG based branch now - where at least some of link-layer security works fine. Will polish and make a PR in near time.

simonduq, 2017-12-14:
That was fast :)

simonduq, 2018-03-26:
Is there any chance we get this ready for 4.1? (i.e. PR in a couple days, before feature-freeze)

simonduq, 2018-03-26:
Assigning milestone 4.2 for now but would be more than happy to upgrade to 4.1!

lafka, 2018-04-30:
@joakimeriksson is the relevant code available anywhere? If you need some help testing i'll give it a roll!

joakimeriksson, 2018-05-01:
Take a look this - this is what I did to try some LLSEC on the CSMA MAC. Not rebased for the latest NG but I guess it could still be useful. Will be rebased and completed before 4.2.

https://github.com/joakimeriksson/contiki-ng/tree/test/llsec-subghz-lwm2m

lafka, 2018-05-01:
Thanks! A few merge conflicts there but other than that I can get basic communicate between two devices almost out of the box (see patch below)

When transmitting a UDP packet between two cc13xx devices the receiver accepts packet and sends an ACK, then sender sees the ACK, ignores it, and does a retransmission. I'll look into it but i'm on thin ice when it comes to the inner working so might come up empty handed.

Any tasks pending before this is merged or just cleaning up? I'll be glad to help out.

Required patch for compiling:
```diff
diff --git a/os/net/mac/llsec802154.h b/os/net/mac/llsec802154.h
index fe1a9d231..af7b3767d 100644
--- a/os/net/mac/llsec802154.h
+++ b/os/net/mac/llsec802154.h
@@ -68,6 +68,12 @@
 #define LLSEC802154_USES_EXPLICIT_KEYS LLSEC802154_ENABLED
 #endif /* LLSEC802154_CONF_USES_EXPLICIT_KEYS */
 
+#ifdef LLSEC802154_CONF_USES_FRAME_COUNTER
+#define LLSEC802154_USES_FRAME_COUNTER LLSEC802154_CONF_USES_FRAME_COUNTER
+#else
+#define LLSEC802154_USES_FRAME_COUNTER LLSEC802154_ENABLED
+#endif /* LLSEC802154_CONF_USES_FRAME_COUNTER */
+
 #ifdef LLSEC802154_CONF_USES_AUX_HEADER
 #define LLSEC802154_USES_AUX_HEADER    LLSEC802154_CONF_USES_AUX_HEADER
 #else /* LLSEC802154_CONF_USES_AUX_HEADER */
```

joakimeriksson, 2018-05-01:
I think there is a few things more to add - specifically related to runtime configuration (changing keys and security levels dynamically).

lafka, 2018-05-03:
Everything is working fine on my side, let me know if there's anything else I can do

simonduq, 2018-05-03:
I do think it's worth opening a PR so we have a good base for discussion :)

joakimeriksson, 2018-05-04:
Sure, but I'll focus on 4.1 release today - then I will open this as a PR - for discussion and improvement. I think we could get it in quite rapidly when 4.1 is out!

lafka, 2018-05-05:
In lack of better place to put this and before i forget:

There's some issue in [rpl-classic/rpl-dag.c:507-508](https://github.com/joakimeriksson/contiki-ng/blob/test/llsec-subghz-lwm2m/os/net/rpl-classic/rpl-dag.c#L507) which breaks router <> dagroot communication.

#### Setup: 
* 1 device acting as DAG root with fd00::/64 and MAC `::a`
* 1 device acting as router with MAC `::b`
* both are compiled with rpl-classic

#### Tests:
1. UDP packet from ::a -> ::b using link-local (fe80::b) works fine
2. UDP packet from ::a -> ::b using prefixed address (fd00::b) works fine
3. UDP packet from ::b -> ::a using link-local (fe80::a) works fine
4. UDP packet from ::b -> ::a using prefixed address (fd00::a) don't work
   * in logs of a `::a`   `[ERR : IPv6      ] invalid version.`

#### Details

In the pcap i can see that communication originating from dag root contain `address compression := Stateless` and `Address Mode := Compressed` in the 6lowpan packet. In communication originating from router `address compression := Stateful` and `Address Mode := Inline`.

Not sure exactly whats going on here but removing those two lines fixes the issue and communication works as normal.

simonduq, 2018-05-06:
Could you please open a new Github issue for this? Thanks!

simonduq, 2018-10-25:
The fix is in https://github.com/contiki-ng/contiki-ng/pull/558 but this will go to `develop`, not 4.2

simonduq, 2019-02-14:
This is ready for 4.3

closed 2019-02-14 by simonduq

MoonPole, 2019-09-25:
Why 'cmd_llsec_setkey' in 'shell-commands' is only for csma?