contiki-ng/contiki-ng #247: Hook or layer between tcpip.c and sicslowpan

laurentderu, 2017-12-13:
In Contiki-NG version of 6LBR, the Packet Forwarding Engine, which is the module that will forward the incoming or outgoing packet toward the right interface and do packet adaptation, is implemented as NETSTACK_NETWORK module.

However, there is still one place where the PFE must be explicitly called : In sicslowpan.c, the code directly invoke tcpip_input() : 

https://github.com/contiki-ng/contiki-ng/blob/d1208f7c5b6bcf3955111f3f7237e624b1fd6833/os/net/ipv6/sicslowpan.c#L1934

This code should instead call the input function of PFE. This could be done by a C callback, or adding a way to chain NETSTACK_NETWORK modules, or create a brand new Layer.

joakimeriksson, 2017-12-13:
Have you looked at the netstack_process_ip_callback function?

laurentderu, 2017-12-13:
I thought the netstack_process_ip_callback was much higher in the stack! Indeed it could do the job nicely.

laurentderu, 2017-12-20:
A bit trickier but it's working eventually.

closed 2017-12-20 by laurentderu