contiki-ng/contiki-ng #248: Replace activation-by-macro of some features by conditional code

laurentderu, 2017-12-13:
Many features in Contiki-NG are only compiled when setting a macro to 1. 6LBR tries to provides runtime configuration of many features and this conflict sometime with the activation-by-macro.

Compiling both Storing and Non-Storing mode in RPL-Classic is not a problem as both can be enabled using the right macro and then the actual mode is selected by the MOP.
On the other hand, features like DAO-Ack or Multicast can not be easily activated at run-time due to how the macro are used.

The proposal is to replace some features activation-by-macro with conditional code and rely on the dead-code removal of the compiler to not increase the code size. For example, the following pseudo code :

```
#if RPL_WITH_DAO_ACK
   // DAO-Ack specific code
#else
  // No DAO-Ack
#endif
``` 

is replaced by :

```
if (RPL_WITH_DAO_ACK) {
   // DAO-Ack specific code
} else {
  // No DAO-Ack
}
``` 

I did the exercice for DAO-Ack, and I got the following figures for Native, Sky and CC2538DK with RPL_CONF_WITH_DAO_ACK set to 0 :

*Native* :
Code : +152 bytes
Data : +0 bytes
BSS : + 32 bytes

*Sky* :
Code : +2 bytes
Data : +2 bytes
BSS : + 20 bytes

*CC2538DK* :
Code : +0 bytes
Data : +4 bytes
BSS : + 32 bytes

(The BSS increase is due to the dao_retransmission_timer in the DAG struct)

When RPL_CONF_WITH_DAO_ACK is set to 1, the code and data size is unchanged.

The full code can be seen here : https://github.com/cetic/contiki-ng/tree/test-dao-ack-compilation

simonduq, 2017-12-13:
What would you think about doing it like `RPL_IS_STORING`? Gets the best of both worlds I believe: enables run-time selection, but still makes it possible to compile the feature out and save memory.

laurentderu, 2017-12-13:
I have to look more closely, For example, in https://github.com/contiki-ng/contiki-ng/blob/9d740ca8e6452ad038f3f7934e2f0810ce54e427/os/net/rpl-classic/rpl-icmp6.c#L1090

it's an #if ... #else ... #endif

You should keep the #if ... #else ... #endif and inside the first block, add a if (RPL_IS_DAO_ACK()) and duplicate the code of the #else branch.

simonduq, 2017-12-13:
We shouldn't need the `#if` at all: `if (RPL_IS_DAO_ACK())` would be enough to (1) check that the feature is included in the binary and if so (2) check if it is run-time enabled.

When not including in the binary, the compiler should take care of removing the always-false branch of the if.

laurentderu, 2017-12-13:
The #if RPL_WITH_DAO_ACK in the rpl_dag_t struct should also be removed then (otherwise the compiler with complain that dao_retransmission_timer does not exist, even if it is not used), and you end up with my first proposal with only a different name for the macro (And the size increase are still valid).

simonduq, 2017-12-13:
Ah ok, right, I think I'm convinced :)