contiki-ng/contiki-ng #249: Add hook in RPL to be notified when DODAG version changes

laurentderu, 2017-12-13:
In 6LBR, we store the current DODAG version in NVM (or config file) in order to directly use the right one when restarting the BR.
The current code in Contiki-NG is to use the lollipop init value and then react when DODAG inconsistency is detected, however this increase the start up time. Also this could lead to configuration inconsistencies if the current DODAG is still using the init value and the BR restarts with a new DODAG configuration and yet still use the init value for the DODAG version.

joakimeriksson, 2017-12-13:
We usually store the DODAG version in the air - e.g. we do send a DIS - take the version and increase - then we are fairly quickly up to a good version and it is not taking too much time either (this is how we handle restarts at Yanzi).

laurentderu, 2017-12-13:
This does not work with deep sleep nodes; and with TSCH, if the nodes has lost the association with the coordinator, it can take some times for them to join again the network and reply to DIS.

joakimeriksson, 2017-12-13:
True - significantly longer time with TSCH.

simonduq, 2017-12-13:
I believe this is very related to https://github.com/contiki-ng/contiki-ng/issues/128

simonduq, 2018-01-16:
As a side note, I believe TSCH actually handles root reboot much more smoothly: nodes lose MAC-level association, notify RPL and leave the DAG, and whenever associating again will promptly re-send DIS and re-join the RPL DAG.

With CSMA, nodes have no such implicit way to find out that there was a reboot.

simonduq, 2018-03-26:
@laurentderu do you have an implem in 6lbr that you want contribute, or would like us to provide the callback?