contiki-ng/contiki-ng #250: Can not have more than one IP address associated with an identical LL address

laurentderu, 2017-12-13:
Currently, it is not possible in the neighbor table to have two IP addresses sharing the same LL address. This occurs however when you communicate with a node using it's link-local IP address and global address (NDP messages often use the link-local address and the application layer the global address)

I made quick workarounds to restore the support, see https://github.com/cetic/contiki-ng/tree/multi-addresses-per-ll

@yatch made some time ago a much better implementation which got never merged into Contiki.

joakimeriksson, 2017-12-13:
I am a bit interested to know a bit more on the specific use-case? We also would have cases when nodes could get IPv6 addresses that are link-local but not similar to the MAC address - e.g. not autoconf created address? (which I think is not all that easy today either?)

laurentderu, 2017-12-13:
This problem does not occurs on RPL network as the communication is always done using the global address, the link-local is only used for broadcasts and so is never registered in the neighbor table.

When using NDP however, the problem occurs usually quickly : two nodes first exchange with their link-local addresses, usually a RS followed by a RA, so in the neighbor table the MAC address is associated with the link-local address. Then, when using the global address to actually communicate, there is an endless stream of NS and NA exchanges. After each NS/NA exchange for the global address, the entry in the neighbor table stays in INCOMPLETE state and so can not be used. As the MAC address was already found and associated with another IP address, the NA message is dropped.

simonduq, 2017-12-14:
@yatch do you want to bring back your old implem to life or let @laurentderu submit the new workaround instead?

yatch, 2017-12-14:
@laurentderu has a good memory! I forgot that work totally... What he refers to must be https://github.com/contiki-os/contiki/pull/2098.

OK, indeed, it is a workaround. https://github.com/contiki-ng/contiki-ng/compare/develop...cetic:multi-addresses-per-ll#diff-dc6d8865c21313ee1ca8b62c9a6d2619R154.

If you could wait, I'd update my work and bring it to contiki-ng. One nicer thing in my work is it has test code :wink:

simonduq, 2017-12-14:
Sounds great @yatch , thanks! Tests are important yes :)

simonduq, 2018-01-16:
@yatch can we reasonably mark this as due for next release? (April)

yatch, 2018-01-16:
@simonduq Yes, we can. Please do that 😄

yatch, 2018-04-01:
I've open a PR for this issue... I'm sorry that it may be too late.

simonduq, 2018-10-19:
Apologies for the lag on that PR. The PR is ready IMO but this is too large a change for last-minute inclusion, I propose merging right after 4.2 feature-freeze, so, for inclusion in 4.3.

closed 2018-10-24 by simonduq