contiki-ng/contiki-ng #252: Early application initialisation

laurentderu, 2017-12-15:
User application is only started when all the platform is initialized and the system configured, however for 6LBR the application must be initialized much earlier as to read the NVM or configuration file and set up the various system configuration parameters which are used in the platform and stack initialization.
Also, on native platform, the link-local address is derived from the MAC address of the slip-radio, which should therefore be retrieved before the stack init.

Currently, this is achieved by disabling the activation of the tcpip_process, which is then started in 6lbr code once the basic initialization has been done.

Three solution here :
* Make the startup of tcpip_process configurable using macro
* Hide the initialization of 6LBR in the init of one of the net stack layer (but this would conflict on no-native platform with the actual radio or mac layer).
* Add explicit hook for early initialization of a user application

g-oikonomou, 2017-12-15:
For MAC addresses, I've been playing around with an API that will populate `linkaddr_node_addr` from hardware values. In the 6lbr example this can then override platform default and retrieve over SLIP.

For the remainder of application-specific configuration, I like the idea of hooks called early in the startup sequence. Perhaps a configuration `#define` that allows the example to tell us what function to call.

lafka, 2018-05-01:
I've opted for the loading configuration from flash in `platform_init_stage_two/0` and then using a combination of macros (see below) and `NETSTACK_RADIO.set_value/2` calls in `platform_init_stage_three/0`.

This works okey in most cases but there are certain parameters (like `IEEE802154_CONF_PANID`) that requires non-constant values. 

The setup is not at all great but it works for my case, if there was a better option to force order of process startup it would be great. My opinion is that an process dependent on TCP/IP (or any other process) should not be started until that process have successfully started (read: your main application). I think the best option would be to allow a user-defined process/function to be called before `netstack_init/0` to handle dynamic configuration or similar.

Example of dynamic configuration constants:

```c
#define RPL_CONF_DIS_INTERVAL   (*((uint16_t *) config_return(CONFIG_RPL_DIS_INTERVAL, NULL)) * CLOCK_SECOND)
```