contiki-ng/contiki-ng #254: List: list_insert bug

simonduq, 2017-12-15:
The function `list_insert` in `list.c` is buggy: when `previtem` is null, it pushes the new element (which (1) removes any old instance and then (2) inserts the new element). But when `previtem` is non-null, it just adds the new item without removing any old instance. Could in duplicate elements in the latter case.

Only reporting as `bug/low` because the function is currently not used in the codebase.

g-oikonomou, 2017-12-15:
I think we need a `list_remove(list, newitem)` at the start of the else block, at line 308

simonduq, 2017-12-15:
Yes I also think this should do

g-oikonomou, 2017-12-15:
I'm looking at this now. The function also assumes that the caller has somehow made sure that `previtem` is already in the list.

g-oikonomou, 2017-12-16:
For the record, things are actually worse than having the same element in the list twice: This bug will corrupt the list.

Imagine you have a list like this:

```
head --> element_1 --> element_2 --> tail
```

Now imagine you want to add the new element to the tail of the list using this function. If you call `list_insert(lst, tail, element_1);`, the function will alter `tail->next` and `element_1->next`, but will _not_ alter `head->next`. The resulting list will be:

```
head --> tail --> element_1
```

PR #255 (2017-12-22): Fix list_insert bug
This PR

* Fixes #254 
* Adds lists to the data structure unit test
* Tidies-up `list.c` a little bit
commit bcd75789 (2017-12-22): Make sure we don't add the same item twice

Fixes #254
closed 2017-12-22 by joakimeriksson