contiki-ng/contiki-ng #259: The build system for platform cooja is all over the place

g-oikonomou, 2017-12-17:
The build system is embarassingly broken for platform cooja.

I suspect it has to do with hacks used to build `cooja.jar` automatically depending on the make goal. Only the logic fails for many goals...

To reproduce, cd into the `examples/hello-world`

* `make distclean` spews out warnings
```
$ make TARGET=cooja distclean
../../Makefile.include:249: warning: overriding commands for target `distclean'
../../arch/platform/cooja/Makefile.cooja:32: warning: ignoring old commands for target `distclean'
rm -f *~ *core core *.srec \
        *.lst *.map \
        *.cprg *.bin *.data contiki*.a *.firmware core-labels.S *.ihex *.ini \
        *.ce *.co
rm -rf 
rm -rf obj_cooja
rm -f hello-world.cooja
```

* `make` fails
```
$ make TARGET=cooja 
make: *** No rule to make target `hello-world', needed by `all'.  Stop.
```

* `make all` fails
```
$ make TARGET=cooja all
make: *** No rule to make target `hello-world', needed by `all'.  Stop.
```

* `make savetarget` spews out warnings, but works
```
$ make TARGET=cooja savetarget
../../Makefile.include:350: warning: overriding commands for target `savetarget'
../../arch/platform/cooja/Makefile.cooja:32: warning: ignoring old commands for target `savetarget'
saving Makefile.target
```
* `make savedefines` spews out warnings, but works
```
$ make TARGET=cooja DEFINES=test savedefines
../../Makefile.include:358: warning: overriding commands for target `savedefines'
../../arch/platform/cooja/Makefile.cooja:32: warning: ignoring old commands for target `savedefines'
saving Makefile.cooja.defines
```

arurke, 2022-12-07:
@pjonsson I randomly came across this - is this issue now moot as we only allow target cooja to be built inside Cooja?

pjonsson, 2022-12-07:
We had a look at this somewhere around the 4.8 release, some issues are fixed but part of what this ticket (most likely) targets was still broken.

There are more simplifications in #2304 that re-enables building outside of Cooja again, but Niclas detected some issue when he tried the PR and I'm a bit swamped before Christmas so I haven't had time to look at it yet.

If we stop symlinking the target binaries in `.`, I think `make distclean` could be implemented as `rm -rf build`, and that would always work.

I have no intention of doing anything with savedefines&friends that are in the build system and are presumably broken. I don't know how it works, and my impression is that using the feature makes it impossible to cut&paste the build command to get an identical build on a different machine.

arurke, 2022-12-07:
Ah, I did not realize #2304 and the intention to bring the building outside of Cooja back - then let's keep this open.

closed 2025-11-19 by joakimeriksson