contiki-ng/contiki-ng #265: Use safer PRI macros in logs

laurentderu, 2017-12-20:
(Follow up of the discussion in #241)

The format placeholders %u, %lu, %h, ... do not have the same meaning depending on the word size of the target platform. When used with fixed size variables, this causes compilation warning on platform which word size is not 32 bits.

For example :

    printf("%lu\n", (uint32_t)1);

will raise a warning on a 64-bits platform, as a long word is 64 bits and not 32.

Two solutions : either uses PRI macros that will select the right placeholder for the target platform, or always do an explicit cast to the related type used in the format string.

joakimeriksson, 2018-01-12:
Yes, we have decided to use the PRI macros for specified size integers. Will add it to the code-style soon.

simonduq, 2018-02-26:
closed by https://github.com/contiki-ng/contiki-ng/pull/297

closed 2018-02-27 by simonduq