contiki-ng/contiki-ng #273: MSP430 issue

iamxg, 2017-12-30:
[image]

iamxg, 2017-12-30:
Hi,
I can't resolved this issue in contiki-os 3.0. I find that there still exist this issue in this version. Who can help me!
Happy new year to everyone!
Xian Guo

simonduq, 2017-12-31:
Thanks for reporting; we should aim for this example to compile even on Sky. Meanwhile, I encourage you to take a look at this: https://github.com/contiki-ng/contiki-ng/wiki/Tutorial:-RAM-and-ROM-usage

g-oikonomou, 2018-01-12:
Yep, this is a bug. We can either document that it won't work, or do something to make it work.

I'm editing the title.

iamxg, 2018-01-12:
Ok, thank you for replying.

Best Regards,

 

 

Xian G

g-oikonomou, 2018-02-02:
The example now documents that it is not expected to work on sky. I'm closing this.

closed 2018-02-02 by g-oikonomou