contiki-ng/contiki-ng #274: Openmote target does not work

laurentderu, 2018-01-05:
Trying several examples from the master and the develop branches for the open mote platform,, I discovered that none of them work for me. The Openmote does not startup at all, it stays in the boot loader mode after flashing the new firmware.

Long story short, it seems that the .vectors section is not located at 0x0000000000200000 as expected but at 0x0000000000202000. The shift is caused by COFFEE which allocate 4 pages at the beginning of the flash on the Openmote, shifting all the other sections. All the examples works perfectly when disabling COFFEE, obviously.

The problem comes from the CCA configuration uses the vectors address before the shift and not the actual vectors address assigned by the linker. (Or my toolchain does not link the firmware properly or I missed something obvious in the documentation). I guess that the Zoul platform has the same problem too, but not tested yet.

g-oikonomou, 2018-01-05:
Zoul works, so this is either a platform thing or a toolchain/local installation thing.

g-oikonomou, 2018-01-05:
Hi Laurent, I just checked again with zoul/firefly and confirmed to work with COFFEE enabled. What are you using to program your openmotes? When using our `.upload` target, the build system invokes cc2538-bsl passing the correct `-a` argument (e.g. `-a 0x00202000`).

laurentderu, 2018-01-05:
Indeed, the problem was between the chair and the keyboard ! I used the upload target on Contiki, but when I switched to NG, I reused a plain cc2538-bsl.py command, without of course the -a... :(

closed 2018-01-05 by laurentderu

IvanIvan12, 2018-03-08:
I am now getting the error:
Target returned: 0x43, Invalid address
when compiling the hello-world like this:
sudo make TARGET=openmote-cc2538 BOARD_REVISION=A1 MOTES=/dev/ttyUSB0 hello-world.upload