contiki-ng/contiki-ng #279: Doxgyen online auto-upload

simonduq, 2018-01-16:
Basically needs to be pushed at
https://github.com/contiki-ng/contiki-ng.github.io/tree/master/ngdoc

whenever a change is merged on `develop`. We could do it in Travis, using secret environment variables for the Github credentials https://docs.travis-ci.com/user/environment-variables/

simonduq, 2018-01-16:
@nvt I believe you looked into this at some point, any more insights you want to share?

nvt, 2018-01-17:
Yes, using Travis might be better than doing something with webhooks. I was looking at the following GitHub Gist notes, which seem to include a complete setup: https://gist.github.com/vidavidorra/548ffbcdae99d752da02#file-auto-deploy_documentation-md

simonduq, 2018-02-18:
Let's assign this one. @nvt would you be willing/able to do this by next release?

nvt, 2018-02-26:
@simonduq Sorry for the late reply! I had another look at the document, and I'm doubting that the access rights for maintainers are sufficient to perform all the different operations involved. It might be work smoother if a repository owner would do it.

simonduq, 2018-02-26:
Well I think push rights to https://github.com/contiki-ng/contiki-ng.github.io is all you should need isn't it?

simonduq, 2018-04-05:
Any update on this?

simonduq, 2018-04-26:
Postponing to 4.2

PR #547 (2018-06-06): Add support for documentation on readthedocs.io
This pull configures Contiki-NG to work on https://readthedocs.org/.

We create a simple landing page which points to various parts of the API documentation. The API documentation itself is an automatically-triggered doxygen build. readthedocs allows us to flag branches/tags as active/inactive. For active ones, pushing triggers a new build which is what we want.

readthedocs will host multiple different versions of the documentation. Currently it only makes sense to show the docs corresponding to branch `develop`, but in the future we can enable  [... cut]
closed 2018-06-06 by simonduq