contiki-ng/contiki-ng #280: The Docker container does not seem to work

LucaZulberti, 2018-01-20:
Hi all, I have followed the wiki to starting up Contiki-NG with docker: [Docker](https://github.com/contiki-ng/contiki-ng/wiki/Docker).

I have pulled down the contiki-ng repository and changed the path in the `contiker` alias to the correct one:
`
alias contiker="docker run --privileged -v $CNG_PATH:/home/luca/remote-git/contiki-ng -e DISPLAY=$DISPLAY -v /tmp/.X11-unix:/tmp/.X11-unix -v /dev/bus/usb:/dev/bus/usb -ti simonduq/contiki-ng"
`

When I run the docker image, the folder `~/contiki-ng` is empty.

Have you got any tips to resolve the problem?

Luca.

PS: The wiki says to put the alias in the `~/.profile` file, but I had to put it in the `./bashrc` file to make the terminal evaluate the alias whenever I open it.

LucaZulberti, 2018-01-20:
I have resolved with this syntax for the `contiker` alias:
`docker run --privileged --mount type=bind,source=/home/luca/remote-git/contiki-ng,destination=/home/user/contiki-ng -e DISPLAY=$DISPLAY -v /tmp/.X11-unix:/tmp/.X11-unix -v /dev/bus/usb:/dev/bus/usb -ti simonduq/contiki-ng`

As the docker-ce documentation states, the use of `--mount` is preferred even if it is more verbose for mounting volumes.
I'm not an expert of Docker, but the documentation states that `--mount` has to be used for bindings.
What we need seems to be a binding between host and container directories and not from a volume to a container directory.

Should `/tmp/.X11-unix` and `/dev/bus/usb` be bound with `--mount type=bind,...` option too?

Luca.

simonduq, 2018-01-22:
Hi,

Thanks for reporting. What is your host OS? This volume/bind thing needs more investigation. From what I gathered, binds are being deprecated, and volumes should be preferred in the future. I do not know what is the best option for USB devices: the `-v` mount was probably only tested with a Linux host. Please keep us updated if you find out more!

Simon

LucaZulberti, 2018-01-22:
My host OS is Ubuntu 18.04 Bionic (development).
I will learn more about docker and I will keep you updated. ;-)

Luca.

LucaZulberti, 2018-01-22:
Ok, the `-v` option works, I have made a mistake..
Volumes are preferred because they are portable and can be managed in several ways, but they are internal to docker. With source code the best option is bind mount because you can keep you source where you want and make it accessible from the host.
Volumes are accessible under `/var/lib/docker/volumes`, but I think it isn't the correct way to use them.

It's better to add a binding from the project directory to the container and use docker to build and flash the contiki-ng OS into the board.
Also the Docker documentation prefer this way to work with source code: [Volumes](https://docs.docker.com/engine/admin/volumes/#good-use-cases-for-bind-mounts).

Luca.

I think the issue can be closed. The use of `-v` or `--mount` is not so relevant for our purposes.

simonduq, 2018-01-22:
Ah great. So, if I got this right, our wiki page does not require any update, or?
If you want to propose enhancements, please take a look at https://github.com/contiki-ng/wiki-sandbox
Thanks!

LucaZulberti, 2018-01-22:
Yes it does not require any update.
Thank you, I will remember it!

closed 2018-01-23 by simonduq