contiki-ng/contiki-ng #285: Contributing OSCORE (OSCOAP) to Contiki-NG/Erbium

Gunzter, 2018-01-29:
Hi all!

SICS Security Lab has a direct asignment from Ericsson to implement [OSCORE](https://github.com/core-wg/oscoap) for several Open Source projects. OSCORE is currently in Working Group Last Call, I have a implementation of the previous OSCORE version based of the "old" Contiki-OS. 

On of those proposed projects is Contiki-NG, I think both OSCORE and Contiki-NG can benefit from each other. Is this of interest to you,Contiki-NG developers and maintainers? 
I have began a design process to update the structure of the OSCORE-Erbium implementation and if you are interested I will gladly share this.

Best Wishes

Martin Gunnarsson

simonduq, 2018-01-29:
Hi Martin,

Yes, this is absolutely wanted! Please note that we have an ongoing rework of CoAP+DTLS in https://github.com/contiki-ng/contiki-ng/pull/167 which should be merged soonish. I would advice basing your port to this PR directly.

Don't hesitate if you need to help. We also have Gitter for more interactive chat https://gitter.im/contiki-ng

Best,
Simon

g-oikonomou, 2018-02-19:
@Gunzter after hearing about OSCORE at EWSN last week, I too am very positive about having it here.

yatch, 2018-06-14:
@Gunzter I'm interested in playing with OSCORE on contiki-ng! Is [oscore_12](https://github.com/Gunzter/contiki-ng/tree/oscore_12) your working branch for this...?

Gunzter, 2018-06-18:
@yatch  Hi! I am glad to hear that you are interested :) Yes oscore_12 is my working branch at the moment. I will soon merge it to the main branch. But the repo you linked is the one I am working on.

yatch, 2018-06-18:
@Gunzter Hi, thank you for your message! I got it 😄

chrysn, 2020-01-21:
> I will soon merge it to the main branch.

Has this been followed up on, or is that still only available in the fork's branch?

AlfHou, 2022-08-28:
Hi! What is the status on this?