contiki-ng/contiki-ng #302: CPU_START_SOURCEFILES variable does not work with *.s assembly files

juliencombattelli, 2018-02-20:
If the CPU_START_SOURCEFILES variable is used to store an assembly file with ".s" extension instead of ".S", the build system does not build any file (except the project source file, hello-world.c in my case).

simonduq, 2018-02-26:
I have no strong opinion whether we should support .s, .S or both. Do you have thoughts about this? If you have a fix for `makefile.include` feel free to suggest as a PR.

juliencombattelli, 2018-02-26:
.S files are already supported by contiki build system. You can see it lines 57, 215, and 267 of the ```makefile.include```. I think it's interesting to keep the support of assembly files, especially if it is planned to add cpu/platforms based on ARM Cortex-M, since their startup files are often written in assembly.

simonduq, 2018-02-26:
Yes we definitely want to keep support for assembly files, the question was more on the file ext we want to support.

juliencombattelli, 2018-02-26:
Changing all .S assembler files is a workable solution. However, it would be more convenient for developers who are porting contiki on new platforms to simply copy and paste this type of files into the source-tree without having to make any changes.

simonduq, 2018-02-26:
Is that what you would have in mind? https://github.com/simonduq/contiki-ng/commit/038cc099f1987195191138126ae78bce70bf0c8d

juliencombattelli, 2018-02-26:
Yes exactly, but the line with the oname variable does not seem very scalable.

simonduq, 2018-02-26:
OK, proposed a slightly improved version at https://github.com/contiki-ng/contiki-ng/pull/321

PR #321 (2018-02-27): Makefile.include: support assembly file extensions .s in addition to .S
Resolves #302
closed 2018-02-27 by simonduq