contiki-ng/contiki-ng #305: Cooja printout incompletely

yoyo890121, 2018-02-23:
Hi, I tried to used `tsch_schedule_print();` to print Orchestra scheduling table, but it printed incompletely in Cooja. 
It works well on cc2538dk, I think it is a Cooja bug.
[image]

It can reproduce in [simple-node example](https://github.com/yoyo890121/contiki-ng/tree/cooja-printout-issue/examples/6tisch/simple-node) and [this setup (BR and a CoAP node)](https://github.com/yoyo890121/contiki-ng/blob/cooja-printout-issue/projects/tsch_schedule.csc)

Thanks!

simonduq, 2018-02-23:
OK, I looked into this, I believe this is indeed on the cooja or cooja mote side, looks like printouts are buffered up to a certain point and then dropped. @nfi any clue where this can be configured?

BTW, https://github.com/contiki-ng/contiki-ng/pull/307 improves a number of things related to this PR.

yatch, 2018-02-23:
I guess it's because of buffer size limitation... I'll take a look this evening.

nfi, 2018-02-23:
Cooja by default has 1 kB buffer to store logs between ticks. If a Contiki node tries to print more than this before it yields back Cooja, the logged data will be truncated.

You could try to increase `MAX_LOG_LENGTH` in `arch/platform/cooja/sys/cooja-log.c` and see if this helps.

yatch, 2018-02-24:
@yoyo890121 Thank you for providing the branch to reproduce the issue. I got the same result as reported with `rpl-tsch-cooja.csc` under `examples/6tisch/simple-node/` on your branch.

As @nfi suggested, `MAX_LOG_LENGTH` is the key. I added the following line into `project-conf.h`  of `examples/6tisch/simple-node/` and got the mote output shown below:
```c
#define MAX_LOG_LENGTH 2048
```
[image]

You may have to do `clean` in order to make sure the change on `MAX_LOG_LENGTH` gets effective.

yoyo890121, 2018-02-24:
OK, it works well.
Thank you!

closed 2018-02-24 by yoyo890121