contiki-ng/contiki-ng #322: Maybe a bug in cc26xx-cc13xx radio driver

zsrkmyn, 2018-02-26:
Hi, we are using commit abd18639b to test our work, but find it's hard for sensors to form a network using RPL-lite.

We dig into the logs and codes, and find something goes wrong [here](https://github.com/contiki-ng/contiki-ng/blob/develop/os/net/mac/csma/csma-output.c#L229). It's seems that `NETSTACK_RADIO.pending_packet()` always returns a number bigger than 0, but the length returned by `NETSTACK_RADIO.read(ackbuf, CSMA_ACK_LEN)` in the following line is always 0. (It's not reasonable, we have packet pending but read a zero-length thing.) So the result of `send_one_packet()` can always be`MAC_TX_COLLISION`, and thus the link status cannot be updated by `link_stats_packet_sent` in `link-stats.c`.

We doubt that there are something wrong with the radio driver, and by reseting `arch/cpu/cc26xx-cc13xx/rf-core/ieee-mode.c` to commit 38987352c, all things go well!

atiselsts, 2018-02-26:
I agree, it's a bug: it cannot receive packets shorter than 5 bytes. `CSMA_ACK_LEN` is 3 bytes.

g-oikonomou, 2018-02-26:
is this in IEEE mode or is it in Prop mode? Did I break things with #294 while trying to fix #293 ?

atiselsts, 2018-02-26:
Oh, I see: `rxConfig.bIncludeCrc` is set to 1, so two byte FCS is also included in the frame and that makes 3+2=5 bytes.

(Do the upper layer expect to get length with the FCS? For example the cc2420 driver does not return it.)

g-oikonomou, 2018-02-26:
Perhaps after #294 we are returning the wrong length? The return value should be pure data payload.

(Which is always why we also need to address #197 ...)

atiselsts, 2018-02-26:
Now I think that `len` is the length without th FCS, so it should be `len < 3` after all.
Look at how `rssi` is extracted from the frame at `len + 2`.

yatch, 2018-02-26:
@atiselsts 
> Look at how rssi is extracted from the frame at len + 2.

Does figure 23-6 of [swcu117h](http://www.ti.com/lit/ug/swcu117h/swcu117h.pdf) help...?

atiselsts, 2018-02-27:
I've given this some thought. Changing to `len < 3` is a no brainer, but...
1) Not sure if the radio driver should enforce this check. While currently the IEEE standards say that CSMA ACK size is 3 bytes, who guarantees that in the future there's no protocol that uses shorter packets? Sending and receiving 2 byte packets works fine on CC2650 - just tested.
2) If the check _should_ be enforced, it should be enforced consistently across all radio drivers and avoiding platform-specific defines or (worse) magic numbers.

g-oikonomou, 2018-02-27:
Yep, it should be 3, not 5. I am flagging this as a bug.

I am happy for the radio driver to skip this check.

The CC2420 driver checks the RX buffer length against <= footer length (which makes sense). This check was somehow carried over to CC2430, CC2530, CC2538, CC2650, only an error in the logic was introduced at some point in the process... For example, the CC2538 driver will check if the buffer length is <= 4 (which seems wrong).

Perhaps what we need to do is simply check `rx_read_entry[8] < 8` (to make sure all expected metadata are in place) and then replace the 8 magic with an automatically calculated number based on the current values of `rfc_CMD_IEEE_RX_t.rxConfig`.

PR #330 (2018-03-02): CC26xx IEEE mode radio driver: allow reception of packets shorter than 5 bytes
Aims to fix #322
closed 2018-03-02 by g-oikonomou