contiki-ng/contiki-ng #328: "Segmentation Fault" in contiki-ng docker container over Docker for Mac

yatch, 2018-02-28:
# Summary

You may get "Segmentation Fault" in a container created over [simonduq/contiki-ng](https://github.com/contiki-ng/contiki-ng/wiki/Docker) image when you're using [Docker for Mac](https://www.docker.com/docker-mac).

```bash
$ docker run -v /Users/yatch/Work/contiki-ng:/home/user/contiki-ng -ti simonduq/contiki-ng
user@eb32057e92c3:~/contiki-ng$ cd tests/14-rpl-lite/
user@eb32057e92c3:~/contiki-ng/tests/14-rpl-lite$ make 01-rpl-up-route.testlog
Running test 01-rpl-up-route with random Seed 1../../tests/simexec.sh: line 67:    13 Segmentation fault      java -Xshare:on -jar $CONTIKI/tools/cooja/dist/cooja.jar -nogui=$CSC -contiki=$CONTIKI -random-seed=$SEED > $BASENAME.$SEED.coojalog
rm: cannot remove 'COOJA.log': No such file or directory
 FAIL
==== 01-rpl-up-route.1.coojalog ====
==== 01-rpl-up-route.1.scriptlog ====
```

Segmentation Fault also is raised by other commands such as `java` or `apt`:
```bash
user@eb32057e92c3:~/contiki-ng/tests/14-rpl-lite$ java -version
Segmentation fault
user@eb32057e92c3:~/contiki-ng/tests/14-rpl-lite$ sudo apt update
Segmentation fault
```

# Workaround

Try it with an image created locally with `tools/docker/Dockerfile`:
```bash
$ docker build -t contiki-ng - < Dockerfile
Sending build context to Docker daemon   5.12kB
Step 1/20 : FROM 32bit/ubuntu:16.04
 ---> cc30c7e0cb5a
Step 2/20 : RUN apt-get update &&   apt-get install -y --no-install-recommends   build-essential doxygen git wge
t unzip python-serial   default-jdk ant srecord iputils-tracepath &&   apt-get clean
 ---> Running in c3d576aaa597
Get:1 http://security.ubuntu.com/ubuntu xenial-security InRelease [102 kB]
...
 ---> Running in 8a727efc634c
Removing intermediate container 8a727efc634c
 ---> 4db85649811b
Successfully built 4db85649811b
Successfully tagged contiki-ng:latest
```

Now you have `contiki-ng` image. Then,

```bash
$ docker run -v /Users/yatch/Work/contiki-ng:/home/user/contiki-ng -ti contiki-ng
user@0eee1cd369b2:~/contiki-ng$ cd tests/14-rpl-lite/
user@0eee1cd369b2:~/contiki-ng/tests/14-rpl-lite$ make 01-rpl-up-route.testlog
Running test 01-rpl-up-route with random Seed 1................................................... OK
```

# Version Information

```bash
$ docker version
Client:
 Version:       17.12.0-ce
 API version:   1.35
 Go version:    go1.9.2
 Git commit:    c97c6d6
 Built: Wed Dec 27 20:03:51 2017
 OS/Arch:       darwin/amd64

Server:
 Engine:
  Version:      17.12.0-ce
  API version:  1.35 (minimum version 1.12)
  Go version:   go1.9.2
  Git commit:   c97c6d6
  Built:        Wed Dec 27 20:12:29 2017
  OS/Arch:      linux/amd64
  Experimental: true

$ docker images simonduq/contiki-ng
REPOSITORY            TAG                 IMAGE ID            CREATED             SIZE
simonduq/contiki-ng   latest              399d22fd7d3e        3 months ago        2.09GB

yatch, 2018-02-28:
This might be my local issue... Please let me know if you need any further information I can collect on my machine...

simonduq, 2018-03-03:
Would be interesting to hear if any other OSX user could use the dockerhub image directly or also had to rebuild. Almost sounds like a Docker for Mac bug.

yatch, 2018-03-03:
`Reset to factory defaults` of "Docker for Mac" resolves this issue!

You can find this command in "Preferences" -> "Reset".

closed 2018-03-03 by yatch