contiki-ng/contiki-ng #331: Will contiki-ng support Content Centric Network(CCN)?

fzy-Line, 2018-03-01:
As we know, CCN is a promising network architecture and contiki is a excellent open source OS. A few years ago, there was a tutorial about CCNx for Contiki, but CCNx is no longer appropriate. Now, CCN-Lite may be a good choice. I have read a conference paper which is a poster for porting ccn-lite to contiki( http://conferences2.sigcomm.org/acm-icn/2016/proceedings/p221-ahlgren.pdf ), but it isn't explian how to port ccn-lite to contiki. I want to know will contiki-ng support CCN or how to port ccn-lite to contiki. thanks!

simonduq, 2018-03-01:
We do not have a current plan for this in the roadmap (https://github.com/contiki-ng/contiki-ng/issues?q=is%3Aopen+is%3Aissue+label%3Aroadmap). If somebody has code to contribute we may consider adding the feature.

fzy-Line, 2018-03-02:
@simonduq Thanks! I will continue to focus on your work and I also expect someone can contribute some codes.

closed 2018-03-02 by simonduq