contiki-ng/contiki-ng #334: Memory corruption in coap-example-client

Gunzter, 2018-03-02:
The coap_endpoint_t server_ep struct gets overwitten after it is populated by coap_parse_endpoint(). I did not spend time looking for where it gets overwritten, but moving "coap_endpoint_t server_ep struct" from line coap-example-client.c:84 to outside the PROCESS_THREAD declaration solves the issue. 

If you run the coap-example-client with "#define LOG_CONF_LEVEL_COAP LOG_LEVEL_DBG" you can see that on line coap-example-client.c:148  "LOG_INFO_COAP_EP(&server_ep);" an incorrect ipv6-address will be printed. 

The issue was found on both native and Cooja-motes.

simonduq, 2018-03-02:
Thanks for reporting, we will investigate

Gunzter, 2018-03-05:
Making the variable static also solves the issue.

g-oikonomou, 2018-03-05:
@Gunzter thanks for reporting this. Feel free to open a pull for this one. The variable can stay where it is (preferable) or it be file-scope, but it needs to be static.

joakimeriksson, 2018-03-05:
I did a quick PR according to the discussions here (static is for sure 100% needed in processes). see #340 .

closed 2018-03-05 by g-oikonomou