contiki-ng/contiki-ng #341: OpenMote target does not work

IvanIvan12, 2018-03-08:
I am getting the error:
Target returned: 0x43, Invalid address
when compiling the hello-world like this:
sudo make TARGET=openmote-cc2538 BOARD_REVISION=A1 MOTES=/dev/ttyUSB0 hello-world.upload

The output is:
  CC        ../../arch/cpu/cc2538/./ieee-addr.c
  CC        ../../arch/cpu/cc2538/cc2538.lds
  CC        ../../arch/cpu/cc2538/./startup-gcc.c
  CC        hello-world.c
  LD        hello-world.elf
arm-none-eabi-objcopy -O binary --gap-fill 0xff hello-world.elf hello-world.bin
python ../../tools/cc2538-bsl/cc2538-bsl.py -e --bootloader-invert-lines -w -v -b 450000 -a 0x00202000 hello-world.bin
Opening port /dev/ttyUSB0, baud 450000
Reading data from hello-world.bin
Cannot auto-detect firmware filetype: Assuming .bin
Connecting to target...
CC2538 PG2.0: 256KB Flash, 32KB SRAM, CCFG at 0x0023FFD4
Primary IEEE Address: 00:12:4B:00:04:30:54:65
Erasing 262144 bytes starting at address 0x00200000
    Erase done
Writing 516096 bytes starting at address 0x00202000
Target returned: 0x43, Invalid address
Target returned: 0x43, Invalid address
...

g-oikonomou, 2018-03-12:
>  sudo make TARGET=openmote-cc2538 BOARD_REVISION=A1 MOTES=/dev/ttyUSB0 hello-world.upload

Can you try without `BOARD_REVISION` and also by replacing `MOTES` with `PORT`. Your console output seems sane so I cannot immediately think of what might be going wrong.

g-oikonomou, 2018-03-12:
The start address of the CC2538 flash is 0x00200000. We start at 0x00202000 because we reserve some pages for CFS. I noticed the erase size too and it made me wonder.

I'll need to dig out my OpenMotes to help any further. Can you try without CFS support?

IvanIvan12, 2018-03-12:
Ok it seems the error was:
BOARD_REVISION=A1

whereas the correct is:
BOARD_REVISION=REV_A1

now the compilation and writing passed without a error:
Erasing 262144 bytes starting at address 0x00200000
    Erase done
Writing 253952 bytes starting at address 0x00202000
Write 248 bytes at 0x0023FF080
    Write done                                
Verifying by comparing CRC32 calculations.
    Verified (match: 0xfa2dd9ec)

IvanIvan12, 2018-03-12:
Sorry for bothering you, I should have seen that earlier.
And thanks a lot for help.

closed 2018-03-12 by g-oikonomou