contiki-ng/contiki-ng #346: How to run the coap example on real devices?

IvanIvan12, 2018-03-13:
I cannot figure out how to run the coap example on real devices and get the data from real devices in Copper. Is it described somewhere how to do that? The README file contains only description for TARGET=native. I have OpenMotes.

chenek, 2018-03-14:
For example, you can use "make TARGET=srf06-cc26xx BOARD=launchpad/cc1310 coap-example-observe-client.bin" to build coap-example-observe-client example running on TI LAUNCHXL-CC1310.

simonduq, 2018-03-14:
We need to update the doc: https://github.com/contiki-ng/contiki-ng/issues/350

closed 2018-03-14 by simonduq

IvanIvan12, 2018-03-14:
@chenek I tried that and it just gives me a lot of errors. But that is a client that it seems by default connects to Cooja, not to the real hardware.