contiki-ng/contiki-ng #349: Why rpl-dag-neighbor module is initialized twice?

fzy-Line, 2018-03-14:
I read the source code of rpl-lite which is a lightweight implementation of RPL. In rpl_init function, rpl_neighbor_init function is  called once.

`void
rpl_init(void)
{
  LOG_INFO("initializing\n");

  /* Initialize multicast address and register it */
  uip_create_linklocal_rplnodes_mcast(&rpl_multicast_addr);
  uip_ds6_maddr_add(&rpl_multicast_addr);

  **rpl_dag_init();
  rpl_neighbor_init();**
  rpl_timers_init();
  rpl_icmp6_init();

  rpl_ns_init();
}`

However, in rpl_dag_init function, the rpl_neighbor_init function is called again.

`void
rpl_dag_init(void)
{
  memset(&curr_instance, 0, sizeof(curr_instance));
  **rpl_neighbor_init();**
}`

May be it's not necessary, or may be it's right but why? Someone can help me?

simonduq, 2018-03-14:
Good catch, this should be called only once. I would only keep the one in `rpl_init`

simonduq, 2018-03-17:
Let me keep it open until we fix it

PR #358 (2018-03-19): Only initialize rpl-neighbor module once
In addition, this PR also fixes potential issues with multiple initialization of nbr-table.

Fixes #349
closed 2018-03-19 by g-oikonomou