contiki-ng/contiki-ng #350: CoAP, DTLS, LWM2M documentation

simonduq, 2018-03-14:
Most of it already exists in the sandbox wiki. We need to update main wiki for the next release.

joakimeriksson, 2018-03-14:
Separat short tutorial on CoAP is probably totally lacking at the moment so that is very much needed. The LWM2M documentation is also on the wiki but needs linking (will fix that on Friday the latest). Then I also have a more or less complete "how-to" tutorial for EFR32 with lots of LWM2M and MQTT. Will try to adapt that to the wiki (and I am still hoping to put up the EFR32 port to Contiki-NG soon).

simonduq, 2018-03-14:
Ah I missed it was there but just not linked :)

simonduq, 2018-04-07:
I added the links to the main page.

One thing that I think is missing is an output example for the LWM2M tutorial. In this case, perhaps a screenshot?

simonduq, 2018-04-13:
CoAP tutorial at Tutorial available at: https://github.com/contiki-ng/contiki-ng/wiki/Tutorial:-CoAP

simonduq, 2018-04-26:
Summary of what's missing:
* [ ] DTLS documentation page
* [ ] Raw DTLS+CoAP tutorial
* [ ] DTLS in LWM2M tutorial

joakimeriksson, 2018-05-04:
Raw DTLS + CoAP will be a bit complex as that will require some external Java or similar stack on the "outside" to test things. Probably we will need the Californium and Scandium for testing it.

simonduq, 2018-05-09:
So, marking at 4.2

simonduq, 2018-10-19:
Update? @joakimeriksson @nfi

simonduq, 2018-10-19:
Only the raw DTLS tutorial is missing. Labeling as long-term.