contiki-ng/contiki-ng #351: RPL border router - no router address in the output

IvanIvan12, 2018-03-14:
According to the "Tutorial: RPL boarder router" I should get the address of the router but the output I get stops before that:
$ sudo make TARGET=openmote-cc2538 BOARD_REVISION=REV_A1 PORT=/dev/ttyUSB0 connect-router
sudo ../../tools/tunslip6 fd00::1/64
********SLIP started on ``/dev/ttyUSB0''
opened tun device ``/dev/tun0''
ifconfig tun0 inet `hostname` mtu 1500 up
ifconfig tun0 add fd00::1/64
ifconfig tun0 add fe80::0:0:0:1/64
ifconfig tun0

tun0      Link encap:UNSPEC  HWaddr 00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00  
          inet addr:172.17.0.2  P-t-P:172.17.0.2  Mask:255.255.255.255
          inet6 addr: fe80::1/64 Scope:Link
          inet6 addr: fe80::dd11:906f:e3b5:27de/64 Scope:Link
          inet6 addr: fd00::1/64 Scope:Global
          UP POINTOPOINT RUNNING NOARP MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:500 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

g-oikonomou, 2018-03-18:
If your device has a reset button, try resetting it with tunslip running.

IvanIvan12, 2018-03-19:
It has reset and user buttons. I tried pressing reset and than even user but unfortunately nothing happens.
I tried pressing the reset button for the hello-world, just to see if it works, and there it seems to restart the program without any problem.

Please say you have some other idea I could try. :)

g-oikonomou, 2018-03-19:
Try increasing tunslip verbosity level

IvanIvan12, 2018-03-19:
Thanks, I increased it to 5. Now I realize that I probably have to upload rpl-boarder-router to the node that I want to be router before starting tunslip? I thought hello-world is enough. It is slightly confusing the tutorial.

Is this the way to do that?
sudo make TARGET=openmote-cc2538 BOARD_REVISION=REV_A1 PORT=/dev/ttyUSB0 MOTES=/dev/ttyUSB0 rpl-border-router.upload

simonduq, 2018-03-19:
Please don't hesitate to suggest a fix to the tutorial for whatever was not clear

IvanIvan12, 2018-03-19:
Thanks, I will as soon as I figure out what was not clear. :)

IvanIvan12, 2018-03-19:
Just please Simon do not close this question jet. You are fast on the trigger. :)

simonduq, 2018-03-19:
Sure. Note that if an issue gets closed too early, you have the option to re-open :)

IvanIvan12, 2018-03-19:
So it is like that, first compile and upload the boarder router something like this:
sudo make TARGET=openmote-cc2538 BOARD_REVISION=REV_A1 PORT=/dev/ttyUSB0 MOTES=/dev/ttyUSB0 border-router.upload

and then run tunslip like it is explained in the tutorial.

Now I get the address but I can not ping it. After:
ping6 fd00::212:4b00:430:5415
I get:
connect: Network is unreachable

but that problem is out of the scope of this question so i guess you can close this one.

IvanIvan12, 2018-03-20:
I still can not ping the router. I run the tunslip within the docker. Is that ok? I guess all the prompts in the tutorial are the prompts of the docker?

I do not understand from where should I ping than the router. I just opened another terminal and ping it but I get the error:
connect: Network is unreachable

What do I need to do?

IvanIvan12, 2018-03-20:
The problem was that I was running tunslip within the docker. Now I can ping the router.
I will definitely have some suggestions for the changes in the documentation.

simonduq, 2018-03-20:
Great. The tutorials were written pre-4.0, that is, without Docker.
Running tunslip6 from inside Docker should also be possible though. Make sure you have the latest Docker (we recently added auto-enabling of IPv6) and Contiker alias. Issue the ping from within the same container that runs tunslip6.

For documentation updates, please either:
* suggest simple fixes as a github issue
* or draft a wiki page on https://github.com/contiki-ng/wiki-sandbox and then open a github issue that points to it

IvanIvan12, 2018-03-20:
Thanks, you can close this one now. :)

simonduq, 2018-03-20:
Thanks for reporting!

closed 2018-03-20 by simonduq