contiki-ng/contiki-ng #352: Build everything script

g-oikonomou, 2018-03-15:
I started fiddling the other day, but I thought I'd open an issue or this may fall through the cracks.

Basically to handle boards the script does this:

```
    if [[ "$platform" == "srf06-cc26xx" ]]
    then
        # srf06-cc26xx has multiple boards
        BOARDS="srf06/cc26xx srf06/cc13xx launchpad/cc2650 launchpad/cc1350 launchpad/cc1310 sensortag/cc2650 sensortag/cc1350"
    elif [[ "$platform" == "zoul" ]]
    then
        # Zoul has multiple boards
        BOARDS="remote-reva remote-revb firefly-reva firefly orion"
    else
        # Other platforms have just a single board
        BOARDS="default"
    fi
```

This means each time we add/remove a board or platform with boards, we have to touch this file. I will unavoidable start getting stale as time passes.

A better way would be to automatically detect the list of boards as we go along. The `boards` make target within hell-world will do it, something like:

```
    # Retrieve the list of supported boards from the vanilla hello-world dir
    TEST_BOARDS=`make -C ../../examples/hello-world TARGET=$platform boards`
```
We may have to slightly tune the output of the boards make target.

By the way, there is also board launchpad/cc1310.

My attempts reminded me why I hate bash scripts...

simonduq, 2018-03-15:
Yeah that sounds good!

g-oikonomou, 2019-02-14:
#700

closed 2019-02-14 by g-oikonomou