contiki-ng/contiki-ng #353: Mote type creation error

algoragit, 2018-03-15:
While trying to run cooja simulations, I get the following error:
 INFO [main] (ContikiMoteType.java:404) - Creating core communicator between Java class Lib1 and Contiki library '/home/CLUSTER/42110030/contiki-ng/projects/rpl-udp/obj_cooja/mtype892.cooja'
 INFO [main] (ContikiMoteType.java:578) - Parsed section at 0x33b60 ( 1768 == 0x6e8 bytes)
 INFO [main] (ContikiMoteType.java:578) - Parsed section at 0x34260 ( 109241 == 0x1aab9 bytes)
FATAL [main] (Cooja.java:1337) - Exception when loading simulation: 
org.contikios.cooja.Cooja$SimulationCreationException: Mote type creation error: Error setting reference variable: Unknown variable name: referenceVar
	at org.contikios.cooja.Cooja.loadSimulationConfig(Cooja.java:3451)
	at org.contikios.cooja.Cooja.loadSimulationConfig(Cooja.java:3361)
	at org.contikios.cooja.Cooja.quickStartSimulationConfig(Cooja.java:1330)
	at org.contikios.cooja.Cooja.main(Cooja.java:3239)
Caused by: org.contikios.cooja.MoteType$MoteTypeCreationException: Error setting reference variable: Unknown variable name: referenceVar
	at org.contikios.cooja.contikimote.ContikiMoteType.doInit(ContikiMoteType.java:492)
	at org.contikios.cooja.contikimote.ContikiMoteType.configureAndInit(ContikiMoteType.java:366)
	at org.contikios.cooja.contikimote.ContikiMoteType.setConfigXML(ContikiMoteType.java:1338)
	at org.contikios.cooja.Simulation.setConfigXML(Simulation.java:703)
	at org.contikios.cooja.Cooja.loadSimulationConfig(Cooja.java:3432)
	... 3 more
Caused by: org.contikios.cooja.mote.memory.UnknownVariableException: Unknown variable name: referenceVar
	at org.contikios.cooja.mote.memory.VarMemory.getVariable(VarMemory.java:104)
	at org.contikios.cooja.contikimote.ContikiMoteType.doInit(ContikiMoteType.java:489)
	... 7 more
Does anyone knows why is this happening?
Best,
Carlos

simonduq, 2018-03-26:
What system are you on? Are you aware of the fact that Cooja motes are only supported on 32-bit linux? See https://github.com/contiki-ng/contiki-ng/issues/134

For now the preferred option to run Cooja motes is Docker https://github.com/contiki-ng/contiki-ng/wiki/Docker

g-oikonomou, 2018-03-29:
I should add that I have seen travis fail transiently with an error output that is similar to the OP. Could be a coincidence.

simonduq, 2018-05-04:
@g-oikonomou you mean the `referenceVar` or the `core communication` thing?
Because the latter should be solved since https://github.com/contiki-ng/cooja/pull/6

There was a "core communication" error that happened all the time and was not fatal. But you wouldn't see it unless you look for it. When a CI test failed, you'd see a dump of the logs, and be easily misled into thinking it was the error that causes the CI failure.

As for this particular issue, not sure we'll be able to investigate anything without more info about the OP's setup.

closed 2020-11-25 by g-oikonomou