contiki-ng/contiki-ng #354: cc13xx platform loads bad MAC address

firmwareguru, 2018-03-17:
The cc13xx platform (specifically observed on launchxl-cc1310 with current develop branch) starts up with a bad MAC address : ff:ff:ff:ff:ff:ff

The problem here is platform.c choosing to load a BLE MAC address because of 
#if MAKE_MAC == MAKE_MAC_BLE
and MAKE_MAC_BLE symbol is not available.

As a quick fix, replacing with this works:
#ifdef MAC_CONF_WITH_BLE

In all cases where the BLE MAC is not specified, these platforms (cc131x, cc135x) should be using the IEEE MAC from the FCFG area.

g-oikonomou, 2018-03-18:
Thanks for the report, will look into asap

g-oikonomou, 2018-03-18:
Confirmed. This is pretty bad.

Fix in #356, @firmwareguru could you confirm please?

firmwareguru, 2018-03-19:
@g-oikonomou  Yes this works, thanks.

PR #356 (2018-03-19): Fix link layer address retrieval from flash
Incorrect pre-processor conditional results in cc13xx/cc26xx devices loading the wrong MAC address.

Fixes #354
closed 2018-03-19 by simonduq