contiki-ng/contiki-ng #359: No sensor data over coap from OpenMote Rev. A1 with OpenUSB Rev. A2

IvanIvan12, 2018-03-20:
I run rpl-boarder router on one of the OpenMotes complied like this:
sudo make TARGET=openmote-cc2538 BOARD_REVISION=REV_A1 PORT=/dev/ttyUSB0 border-router.upload

and tunslip:
sudo make TARGET=openmote-cc2538 BOARD_REVISION=REV_A1 PORT=/dev/ttyUSB0 connect-router

On another mote I run coap serer like this:
sudo make TARGET=openmote-cc2538 BOARD_REVISION=REV_A1 PORT=/dev/ttyUSB1 coap-example-server.upload 

The problem is that in the cooper i do not see any sensors:
[image]

simonduq, 2018-03-26:
Thanks for reporting. The reason the resources do not show up is because flags such as `PLATFORM_HAS_TEMPERATURE` etc. are not set on this platform. We do support temperature sensing on this platform, but not through the interface expected in this example. That's a problem.

@g-oikonomou should we add "sensors" to the list in https://github.com/contiki-ng/contiki-ng/issues/77 ?

IvanIvan12, 2018-03-26:
Thanks.
Interestingly, the person who sells OpenMotes, referred me to contiki-ng because I was not able to get the same example working on OpenWSN, since the OpenWSN is not ready for it according to his words. If I only knew, I would have never bought the device.

I guess you can close this issue and thanks once again.

g-oikonomou, 2018-03-26:
I think sensors are an entirely different can of worms, probably one that deserves its own issue!

closed 2018-03-26 by g-oikonomou

g-oikonomou, 2018-03-26:
I am looking at this again. Setting `PLATFORM_HAS_TEMPERATURE` is one thing. The second thing is to create a sensor variable named `temperature_sensor` and get it to use the existing sht21 driver. Unless I have missed something, it should be a trivial diff to make this work for the temperature sensor at least.

g-oikonomou, 2018-03-26:
Ditto light using the max44009, I think

simonduq, 2018-03-26:
Yes but then we'd want not only temperature and not only this paltform. Opening new issue.

g-oikonomou, 2018-03-26:
Yes sensors API is one thing. My subsequent posts were about getting some OpenMote sensor support in this example using the current code base

simonduq, 2018-03-26:
Agree, @IvanIvan12 getting this particular example to work on OpenMote should be quite trivial.
As for the HAL, new issue at https://github.com/contiki-ng/contiki-ng/issues/374

IvanIvan12, 2018-03-29:
Thanks. I managed to do it. It was not too hard thanks to your instructions.

simonduq, 2018-03-29:
Awesome :)

g-oikonomou, 2018-03-30:
@IvanIvan12 feel free to open a pull request for this one

IvanIvan12, 2018-04-03:
Thanks, I will.