contiki-ng/contiki-ng #374: HAL for sensors

simonduq, 2018-03-26:
We need to homogenize access to e.g. temperature sensor. There exists some very minimal HAL but it's only supported by a small subset of platforms. Consists basically in `PLATFORM_HAS_TEMPERATURE` and `temperature_sensor`.

Related to https://github.com/contiki-ng/contiki-ng/issues/359

g-oikonomou, 2018-03-26:
I've been thinking about this for over a year...

We need a HAL that:
* Acknowledges the fact that a sensor (sensing element / hardware part) can support multiple sensor readings (e.g. an SHT can give you temperature and humidity).
* Supports asynchronous readings (some sensors can take hundreds of ms from power up to reading ready). A uniform way to do this with events would be nice hear, see some of the sensortag sensors.
* Allows each sensing element to identify itself (e.g. I can provide temperature and humidity readings).
* Allows each reading to provide some meta data about itself, e.g. the unit of the reading.

When it comes to cross-platform examples, I am not sure whether we want to keep the `PLATFORM_HAS_xyz` approach (advantage: reduces code size) or switch to a more generic approach whereby an example simply understands the concepts of "sensing element" and "sensor reading" and treats them all the same way irrespective of whether it's a temperature/light/barometer/whathaveyou.

g-oikonomou, 2018-04-02:
Hi folks, I've started knocking something together for this one. Please post with requirements.

joakimeriksson, 2018-04-03:
Another good thing would be to support multiple instances of the same sensor reading type. At least multiple temperature sensors coming from different chips. And even better support multiple instances with the same chip (e.g. having two sht11 or similar).

g-oikonomou, 2018-04-03:
> Another good thing would be to support multiple instances of the same sensor reading type

In which scenario, what do we want to be able to do? What's the use case?

arurke, 2018-04-27:
+1 on async readings and the multiple sensors in one driver (we hijacked various parameters in `sensors_sensor`, similar to what is done for mpu-9250).

Another point:
* A uniform way to represent the current state of the sensor (e.g. initializing, taking reading, reading ready etc. (sensors.h has a few))

joakimeriksson, 2018-06-03:
Many of the current devboards have several instances of temperature in different sensors so I do think we should allow reading all the different temperatures