contiki-ng/contiki-ng #386: Slip Radio sending duplicate messages

beleyj, 2018-03-27:
So I have been running tests on a CoAP testbed set up and everything has been running perfectly. Here is a overview of it:

[image]

However, I just used Sensniff to monitor traffic coming from the 802.15.4 SLIP Radio to the CoAP Server (Ti SensorTag). For every single request that my CoAP client sends, the SLIP radio takes that single request (which originates on IPv4) and sends it 10-20 times over 802.15.4. The CoAP server responds with one message, just like it suppose to.

Does anyone know why the SLIP radio is sending the single request so many times? When I examined the "duplicate request packets", the last byte in the data section does change, but think that means anything. Additionally every packet has a bad check sum, BAD FCS.

Does the border router and the SLIP radio have a bug or is there another explanation for this? 

Wireshark Output:
SLIP Radio IP : 00:12:4b:00:12:05:06:30	   CoAP Server IP:  00:12:4b:00:12:04:ec:7
[Wireshark_output.zip](https://github.com/contiki-ng/contiki-ng/files/1853898/Wireshark_output.zip)

Thanks for the help!

g-oikonomou, 2018-03-27:
Hi. Interesting. Had you been using ContikiMAC I'd have said that this is perfectly normal, but we don't have ContikiMAC here. What did you use to build your SLIP radio? Presumably you are using CSMA MAC on the CoAP server side of things?

There is no bad FCS in any of the frames of the packet capture, which means you need to adjust the configuration of your Wireshark. Use TI FCS under .15.4 preferences and set `fd00::/64` under "Context 0" in 6LoWPAN preferences.

If you can provide the SHAs of the version that you used to build slip-radio and the CoAP server I might be able to assist more.

beleyj, 2018-03-27:
Thanks for the quick reply. I just checked and realized I am using an older .bin of the SLIP radio example from Contiki 3.0 and not from Contiki-NG. Therefore, it is using ContikiMAC and explains this situation. Thank you for the help.

Why in Contiki-NG did you decide to remove support for ContikiMAC?

g-oikonomou, 2018-03-27:
ContikiMAC is not standard. We focus on standards. CSL is in the wishlist :)

simonduq, 2018-03-28:
OK, closing then, @beleyj let us know if you want us to re-open

closed 2018-03-28 by simonduq