contiki-ng/contiki-ng #389: cc26xx-web-demo doesn't work

janeksz, 2018-03-28:
tested on cc2650SensorTag keeps blinking, doesn't join the network http://e2e.ti.com/support/wireless_connectivity/zigbee_6lowpan_802-15-4_mac/f/158/p/673650/2487357#2487357

g-oikonomou, 2018-03-28:
Are the Contiki-NG and 6lbr versions that you are using known to work with each other? Have you checked this page out? https://github.com/cetic/6lbr/wiki/Contiki-NG

janeksz, 2018-03-28:
Thanks, George
I had a naive idea, that old Contiki, and Contiki NG are compatible and I didn't recompile 6lbr on my RPi.
I created a simple network with a 6lbr edge router running on RPi and a few CC2650 SensorTag motes running old Contiki examples (cc26xx-web-demo). Next, I built and flashed another SensorTag with a new ContikiNG image of cc26xx-web-demo. I see it doesn't and will not work. Thanks for clarifying it. Honestly, it's disappointing, but has to be accepted.
Cheers,
Jan

closed 2018-03-28 by g-oikonomou

simonduq, 2018-03-29:
Unless you need specific features of 6lbr you might want to try the border router from main repo: https://github.com/contiki-ng/contiki-ng/wiki/Tutorial:-RPL-border-router

janeksz, 2018-04-06:
thanks, appreciate it