contiki-ng/contiki-ng #390: Update "Platform srf06 cc26xx" wiki page to make it more explicit/entry friendly

Jappedonquixote, 2018-03-28:
I have been tinkering around with the srf06 cc26xx platform in combination with a CC2650EM for a little while now (macos), and in the process started over with a "clean slate" a couple times. The first time I had problems getting the hello-world example run out of the box, and while starting over each timea ran into the same issues, and solved them eventually using some tutorials/forum threads found randomly online.

To my best knowledge these are missing in the Wiki or could be more explicit written down to make life easier for new users. I am not sure to which extent this only applies to my specific board or for every one using the "Platform srf06 cc26xx" wiki page. Also if this has already been accounted for, but I lack this knowledge.

Starting from scratch you basically use the following Wiki pages in succesion to each other:

https://github.com/contiki-ng/contiki-ng/wiki/Toolchain-installation-on-OS-X
https://github.com/contiki-ng/contiki-ng/wiki/Platform-srf06-cc26xx
https://github.com/contiki-ng/contiki-ng/wiki/Tutorial:-Hello,-World!

Everytime while following I would at least once run into a "Timeout waiting for ACK/NACK after 'Get Status (0x23)" error.

I think this is mainly due to three things:

1. In https://github.com/contiki-ng/contiki-ng/wiki/Platform-srf06-cc26xx it is stated as a second condition that:

"The chip is not programmed with a valid image, or the image has the bootloader backdoor unlocked. To enable the bootloader backdoor in your image, define ROM_BOOTLOADER_ENABLE to 1 in contiki-conf.h."

Stated as a second point it can be read over pretty easily, missing important information. 

2. The text:
"The chip is not programmed with a valid image, or the image has the bootloader backdoor unlocked. To enable the bootloader backdoor in your image, define ROM_BOOTLOADER_ENABLE to 1 in contiki-conf.h." 

also implictily tries to tell you that you have to make sure that either there is no image on the board or that the image currently on the board already has the bootloader backdoor unlocked. This is for most people propably not the case, and in order to make it work it is required to upload an image that enables the bootloader using a flashing program to the board before the provided script will ever work. I my case I used Uniflash for this (http://processors.wiki.ti.com/index.php/Category:CCS_UniFlash).

3. Apart from setting define ROM_BOOTLOADER_ENABLE to 1 in contiki-ng/arch/cpu/cc26xx-cc13xx/cc13xx-cc26xx-conf.h and not in contiki-conf.h." (is this still an existing file?) as the Wiki now states also the following has to be set which by my knowledge is not yet documented anywhere else:

**in arch/cpu/cc26xx-cc13xx/lib/cc26xxware/startup_files/ccfg.c lines 179 to 197**: 
#define SET_CCFG_BL_CONFIG_BOOTLOADER_ENABLE 0xC5       // Enable ROM boot loader
#define SET_CCFG_BL_CONFIG_BL_LEVEL                0x0        // Active low to open boot loader backdoor
#define SET_CCFG_BL_CONFIG_BL_ENABLE             0xC5     // Enabled boot loader backdoor 

the correct lines are already there but by default, but still commented out.

**Proposed Fix**
I would suggest editing the "Platform srf06 cc26xx" wiki page by addig a subsection under "How to program your device" called "Ability to enter the boot loader by enabling the bootloader backdoor":

*Ability to enter the boot loader by enabling the bootloader backdoor*
To make sure you can program your board using the serial uploader script, the image on the chip needs to contain an image with has the bootloader backdoor unlocked. Most likely this is not the case by default in which case the following steps can be taken:

* Set: 
#define ROM_BOOTLOADER_ENABLE to 1 in contiki-ng/arch/cpu/cc26xx-cc13xx/cc13xx-cc26xx-conf.h
* Make sure the following is values are set correctly in arch/cpu/cc26xx-cc13xx/lib/cc26xxware/startup_files/ccfg.c lines 179 to 197**:
#define SET_CCFG_BL_CONFIG_BOOTLOADER_ENABLE 0xC5       // Enable ROM boot loader
#define SET_CCFG_BL_CONFIG_BL_LEVEL                0x0        // Active low to open boot loader backdoor
#define SET_CCFG_BL_CONFIG_BL_ENABLE             0xC5     // Enabled boot loader backdoor
* To make sure a correct image is on the chip in case the serial uploader script is not working TI's UniFlash tool can be used for the first upload (http://processors.wiki.ti.com/index.php/Category:CCS_UniFlash) by selecting the correct board and upload the compiled .bin file as an image.
* For uploads on boards that already contain an image with has the bootloader backdoor unlocked you can then enter the boot loader and use the .upload make target (e.g. make cc26xx-demo.upload for the cc26xx-demo). On the SmartRF06, you enter the boot loader by resetting the EM (EM RESET button) while holding the select button. For the LaunchPad, you enter the bootloader by resetting the chip while holding BTN_1. The serial uploader script will automatically pick the first available serial port. If this is not the port where your node is connected, you can force the script to use a specific port by defining the PORT argument eg. make cc26xx-demo.upload PORT=/dev/tty.usbserial.

It is possible to change the pin and its corresponding level (High/Low) that will trigger bootloader mode by changing SET_CCFG_BL_CONFIG_BL_LEVEL and SET_CCFG_BL_CONFIG_BL_PIN_NUMBER in board.h.

For more information on the serial bootloader, the README under the tools/cc2538-bsl directory provides detailed information.

Jappedonquixote, 2018-03-28:
I am aware that this is a lot of text, but I hope this makes it more clear. Myabe an alternative might be to change the code in such a way that when #define ROM_BOOTLOADER_ENABLE 1 is set, automatically also:

#define SET_CCFG_BL_CONFIG_BOOTLOADER_ENABLE 0xC5 // Enable ROM boot loader
#define SET_CCFG_BL_CONFIG_BL_LEVEL 0x0 // Active low to open boot loader backdoor
#define SET_CCFG_BL_CONFIG_BL_ENABLE 0xC5 // Enabled boot loader backdoor

are defined like this.

g-oikonomou, 2018-03-28:
Thanks for the input, very valuable. I went through it, but obviously a fair bit to digest!

g-oikonomou, 2018-03-29:
Hi @Jappedonquixote this was very useful

> Apart from setting define ROM_BOOTLOADER_ENABLE to 1 in contiki-ng/arch/cpu/cc26xx-cc13xx/cc13xx-cc26xx-conf.h and not in contiki-conf.h."

You are of course right. I have changed this.

> in arch/cpu/cc26xx-cc13xx/lib/cc26xxware/startup_files/ccfg.c lines 179 to 197:
#define SET_CCFG_BL_CONFIG_BOOTLOADER_ENABLE 0xC5 // Enable ROM boot loader
#define SET_CCFG_BL_CONFIG_BL_LEVEL 0x0 // Active low to open boot loader backdoor
#define SET_CCFG_BL_CONFIG_BL_ENABLE 0xC5 // Enabled boot loader backdoor

It appears as though you may be using an old version of CC26xxware (or CC13xxware). In our current version, all those lines are conditional, like so:

```
#ifndef SET_CCFG_BL_CONFIG_BOOTLOADER_ENABLE
#define SET_CCFG_BL_CONFIG_BOOTLOADER_ENABLE            0x00       // Disable ROM boot loader
// #define SET_CCFG_BL_CONFIG_BOOTLOADER_ENABLE         0xC5       // Enable ROM boot loader
#endif
```
Setting `ROM_BOOTLOADER_ENABLE` to 1 will automatically set correct values for all those defines through the corresponding board's `board.h`, for example in `arch/platform/srf06-cc26xx/launchpad/cc2650/board.h` we do this:

```
#if ROM_BOOTLOADER_ENABLE
#define SET_CCFG_BL_CONFIG_BOOTLOADER_ENABLE            0xC5
#define SET_CCFG_BL_CONFIG_BL_LEVEL                     0x00
#define SET_CCFG_BL_CONFIG_BL_PIN_NUMBER                BOARD_IOID_KEY_LEFT
#define SET_CCFG_BL_CONFIG_BL_ENABLE                    0xC5
#else
#define SET_CCFG_BL_CONFIG_BOOTLOADER_ENABLE            0x00
#define SET_CCFG_BL_CONFIG_BL_LEVEL                     0x01
#define SET_CCFG_BL_CONFIG_BL_PIN_NUMBER                0xFF
#define SET_CCFG_BL_CONFIG_BL_ENABLE                    0xFF
#endif
```

This enables the ROM bootloader and automatically selects the correct pin and the level required to enter the bootloader.

> I would suggest editing the "Platform srf06 cc26xx" wiki page by addig a subsection under "How to program your device" called "Ability to enter the boot loader by enabling the bootloader backdoor":

OK makes sense. Where in this page do you think such section would make the most sense?

> To make sure a correct image is on the chip in case the serial uploader script is not working TI's UniFlash tool can be used for the first upload

Yep, again makes sense. Or one can use SmartRF flash programmer. In either case, one can simply erase flash (not necessary to do the first programming using that tool).

> For uploads on boards that already contain an image with has the bootloader backdoor unlocked you can then enter the boot loader and use the .upload make target 

Yes, this makes sense too.

g-oikonomou, 2018-03-29:
I have reworked this page a bit based on your feedback

* Under "Requirements", I have added:

    For additional help on how to set your system up, you may also find the guides below useful:

    * Native toolchain installation (Linux)
    * Native toolchain installation (OS X)

* I have sub-sectioned "How to Program your Device". We now have information on how to do things over JTAG and separate information about using the ROM bootloader. In the JTAG section, I have added information about UniFlash.

* I have done some re-writing parts of the ROM bootloader section (I want to do some more).

Thoughts welcome

Jappedonquixote, 2018-04-03:
Thanks for the comment. I was super confused as to why my files were containing the old code, but now rechecking it using the new code everything seems to work quite smooth using the ifdefs. I think the sub-sectioning of "How to program your device" will help a lot. I would still maybe change the line "(see previous subsection)" into (see subsection over JTAG) for sake of explicitness.

Jappedonquixote, 2018-04-03:
Altough I still do not get why the Bootloader is not set to 1 by default since this seems the case for the other platforms?

g-oikonomou, 2018-04-28:
I have updated the wiki to what is hopefully much clearer. I am also changing the default to enable the ROM bootloader. This issue will be closed when #473 is merged.

Thanks for the feedback.

PR #473 (2018-04-28): Enable the CC13x0/CC26x0 ROM bootloader by default
After having locked myself out of my kit for the gazzillionth time and after multiple complaints by our users, I have decided to change the CC13x0/CC26x0 ROM bootloader default to 'enabled'.

But I have also updated the wiki page with clear instructions that it needs disabled for production/deployment images...

Closes #390
closed 2018-04-28 by g-oikonomou