contiki-ng/contiki-ng #395: Platforms overwrite each other's build output

g-oikonomou, 2018-03-30:
Let's take hello-world as an example, but this is the case for all our examples at the moment.

* `make TARGET=zoul` results in a `hello-world.bin` file.
* `make TARGET=nrf52dk` results in a different `hello-world.bin` file, silently overwriting the zoul one.

This poses a risk of a user programming one device with a firmware built for a different device. This is the case for multiple platforms, not just those two. This is the case for other build outputs too, e.g. the `.elf` and `.hex`.

Options:
* It's fine, ignore it
* Include the platform's name in the filename. Perhaps even the board's name too
* Create `.bin` (and friends) in a subdir

IMHO this is a bug, but I'll just flag it as discussion for now.

Thoughts?

yatch, 2018-03-31:
I'm not sure whether it's a bug or not... How about putting all the resulting files other than `.$(TARGET)` into `obj_$(TARGET)`?

g-oikonomou, 2019-02-14:
Closed by #698

closed 2019-03-02 by g-oikonomou