contiki-ng/contiki-ng #409: switch-case indent in uncrustify config missing

zhitaoh, 2018-04-03:
The [code style](https://github.com/contiki-ng/contiki-ng/wiki/Code-style) says:

>  Indent "case" statement and "default" label by two spaces in "switch" statement.

And provides the following example:
```c
  switch(c) {
    case 19:
      return 1;
    default:
      break;
  }
```

The current [configuration](https://github.com/contiki-ng/contiki-ng/blob/develop/tools/code-style/uncrustify.cfg) of the uncrustify beautifier does not enforce this rule. Should this rule be removed from the code style, or should the rule be added to the uncrustify config? If the latter, I find adding the follow line would do the job:

`indent_switch_case = 2`

g-oikonomou, 2018-04-03:
I would rather change the code style example to be honest, most of the code currently does what uncrustify does (probably not a coincidence... :))

zhitaoh, 2018-04-05:
I agree to remove the example from the Wiki page and keep the uncrustify rules. Here is the change:
[https://github.com/contiki-ng/wiki-sandbox/wiki/Code-style](https://github.com/contiki-ng/wiki-sandbox/wiki/Code-style)

g-oikonomou, 2018-04-05:
I am now wondering whether we should change the example to match what uncrustify does, instead of entirely removing. Thoughts?

nfi, 2018-04-05:
For code style, I think more code examples are preferred over less. I would also rather keep the example but changing it to match uncrustify.

zhitaoh, 2018-04-06:
Swtich-case example restored, without inner indentation:
[https://github.com/contiki-ng/wiki-sandbox/wiki/Code-style](https://github.com/contiki-ng/wiki-sandbox/wiki/Code-style)

g-oikonomou, 2018-04-23:
@simonduq do you have a chance to look at this?