contiki-ng/contiki-ng #420: CoAP documentation and tutorials

simonduq, 2018-04-07:
The CoAP documentation page mentions more doc to come after #16 and #42 is merged... which is now the case :) Also, we're still lacking a CoAP tutorial.

I'm happy to take care of creating the tutorial.
@nfi @joakimeriksson for the doc?

zhitaoh, 2018-04-10:
The current examples/coap dir contains an example client that uses the blocking API. Ideally there should be a separate example that showcases the usage of the alternative non-blocking, "callback" API.

It would be good to have another example to show how to switch on transport layer encryption using the coap_endpoint_connect()/disconnect() functions.

simonduq, 2018-04-13:
Tutorial available at: https://github.com/contiki-ng/contiki-ng/wiki/Tutorial:-CoAP

simonduq, 2018-04-26:
And documentation at https://github.com/contiki-ng/contiki-ng/wiki/Documentation:-CoAP

closed 2018-04-26 by simonduq